feelfreelinux/octo4a #2: Hi be gentle i dont know how to use git hub

benjjyman, 2021-03-29:
Hi like I said this the first time I've used gothub for anything but downloading.
Just wanted to say thanks for this app I have so many obselete android boxes this would be a perfect use for them.
So my question is can it be run on a tablet? And can it be run on a dodge Chinese TV streaming device, rooted 2g ram hdmi WiFi, 3 USB ports duel and quad core running android 7+, these things can be bought for less than £20. Would cool if it works.

feelfreelinux, 2021-03-29:
> Hi like I said this the first time I've used gothub for anything but downloading.
> Just wanted to say thanks for this app I have so many obselete android boxes this would be a perfect use for them. [...]

Hi, yes the app should run on any android 5.0+ device that supports OTG. I've also confirmed that the app works well on cheap android TV boxes. Someone even got it running on a cheapest 15$ rockchip tv box.

Thanks for testing my app.

feelfreelinux, 2021-04-01:
Closed as its more of discusison - not really an issue

closed 2021-04-01 by feelfreelinux