feelfreelinux/octo4a #3: usb c hub issues

eppitti, 2021-04-01:
hello I managed to configure the application and make it work without problems on a mia1 and a mi4c, the only inconvenience is a problem with the device charging : to power the device and connect the printer I am using a usb c hub with power delivery function but I can't get both the battery charging and the printer connection to work. What do you recommend to replace it with?

alufers, 2021-04-01:
Hi!
You could try using a simple USB Y cable (example: http://pl.aliexpress.com/item/4000383308064.html ). It is much simpler than a USB PD hub, so there is a chance it will be compatible with your phone.

Unfortunately it is also possible that your phone does not support charging and OTG at the same time at all. Sometimes this is a software limitation and it is possible to lift it by changing your ROM. 

If changing your ROM does not help, the only option is hardwiring the battery connector to 3.7V. But this is a bit dangerous and you need some skill to do this.

closed 2021-04-01 by alufers

JamesNewton, 2021-11-28:
I've run into this problem several times now. I've made a (non-exhaustive) list of the phones I know will support charge and OTG:
http://techref.massmind.org/techref/io/usb/OTG.htm