feelfreelinux/octo4a #10: Request to add Biqu B1 printer to USB Driver

Nha-Krul, 2021-04-12:
First I want to thank you for the grate work that you are doing.
I want to ask if you can add the Biqu B1 printer to de USB Driver.
VID: 0x1D50 (OpenMoko, Inc.)
PID: 0x6029 (Marlin 2.0 (Serial))

Thanks

feelfreelinux, 2021-04-13:
Will add the printer with a new update, thanks for letting me know :)

SamuelRilling, 2021-04-15:
I absolutely love the project!
However i can't connect due to the "Printer not connected" error on my Anet A8, wonder if that could be implemented on a future update as well. Thanks in advance, keep up the good work!

darranwil, 2021-04-27:
Thank you as well for this great package. I also would request the previously listed VID/PID be added.

feelfreelinux, 2021-09-29:
Added in 1.0.0 :) https://github.com/feelfreelinux/octo4a/releases/tag/1.0.0

closed 2021-09-29 by feelfreelinux