feelfreelinux/octo4a #12: 在电视盒上使用，每次开机都跳出来需要授权访问USB

dgpower, 2021-05-15:


dgpower, 2021-05-15:
我用H96 pro TV BOX安装了你的app，并设定为开机自动启动，这样我就可以脱离电视机屏幕，让app自动工作。
但是，每次开机，app都会跳出一个窗口出来询问：是否允许访问USB设备。这导致了我必须接上电视机和键盘来处理这个问题。

I installed your app with H96 pro TV BOX and set it to start automatically after booting, so that I can leave the TV screen and let the app work automatically.
However, every time you boot up, the app will pop up a window asking whether to allow access to the USB device. This resulted in me having to connect the TV and keyboard to deal with this problem.

ferrets6, 2021-06-12:
I had the same problem.
As a workaround, I use Automate with this flow

[image]

1. Start octo4a
2. Inspect layout change
[image]
[image]
In this step, Automate wait for the SystemUI popup to show
3. Click the "Ok" button
[image]
[image]

If you don't know Automate, it's a very powerfull app!

xiongyanbo, 2021-08-23:
> 我用H96 pro TV BOX安装了你的app，并设定为开机自动启动，这样我就可以脱离电视机屏幕，让app自动工作。
> 但是，每次开机，app都会跳出一个窗口出来询问：是否允许访问USB设备。这导致了我必须接上电视机和键盘来处理这个问题。
>
> I installed your app with H96 pro TV BOX and set it to start automatically [...]

拔插一下usb有惊喜。
略微有点烦而已
我是用tasker做了自动点击。
最不好用的是上传文件会报错。实际是正常的。联机上传gcode会报500错误。实际上传成功。不会自动开始打印

feelfreelinux, 2021-11-08:
Rewritten usb driver + auto boot option released in octo4a 1.0.0

closed 2021-11-08 by feelfreelinux