feelfreelinux/octo4a #16: Standalone camera server

hungrierwtf, 2021-06-06:
Unfortunately my Android device doesn't support OTG host and charging at the same time, so I can't use it for my main OctoPrint server. However, I've found that I can use the built-in camera server with my existing OctoPrint instance on another device, which works great because my phone has a pretty good camera and flash for lighting.

Would it be possible to release the camera server as a standalone app for a use case like mine, where I don't need the rest of OctoPrint running on the device? Out of all the Android MJPG stream servers I've been able to find, the included one here works the best for me.

ferrets6, 2021-06-08:
If can be helpfull, there are some other app on Play Store to do this
I use IP Webcam by Pavel Khlebovich on my octo4a

demuthrc, 2021-07-09:
I had the same issue, I've rigged up an old wireless charger and that solved my issues.  Not sure if that is an option for you or not.

feelfreelinux, 2021-11-08:
Currently I have no plans in separating the camera server from the main app, as the camera server itself lacks a lot of features. As @ferrets6 mentioned, [IP Webcam](https://play.google.com/store/apps/details?id=com.pas.webcam&hl=en&gl=US) is a great pick for a standalone camera.

closed 2021-11-08 by feelfreelinux