feelfreelinux/octo4a #18: [Request] Remove F-Droid Antifeatures

seniorm0ment, 2021-06-10:
AntiFeatures: Tracks or reports your activity

feelfreelinux, 2021-06-13:
Hi! Does making tracking opt in / opt out still count as F-Droid antifeature? For now, having bugsnag analysis really helps with development

seniorm0ment, 2021-06-13:
https://f-droid.org/wiki/page/AntiFeatures
Here are the antifeatures docs, I cannot confirm off hand but you may want to ask in the fdroid chat

S0und, 2021-06-23:
@seniorm0ment  feel free to create your own fork, remove and remove it

seniorm0ment, 2021-06-23:
creating a fork just to remove antifeatures, then submit it to fdroid is excessive imo. Also there'd end up being zero reason for users to choose @feelfreelinux's branch over mine in that case.
also I don't actually care that much, I'd rather just not use the app in that case quite honestly. the app isn't something (I) need, but just nice to have yk.

feelfreelinux, 2021-11-03:
Made bugreporting opt-in in https://github.com/feelfreelinux/octo4a/commit/46b00618d59cb1481a0448212e432515e505a52d

closed 2021-11-03 by feelfreelinux