feelfreelinux/octo4a #19: Correct SSH Credentials

man2607, 2021-06-14:
Hi, 
I am trying to ssh into the server, but I keep getting "connection refused" errors. It isn't working when I use the username I setup in octoprint using the browser either. From the app I can only see and set the ssh password. Kindly point me in the correct direction regarding this. 
Thanks!

feelfreelinux, 2021-06-15:
Hi! I really should improve the ssh documentation :) In 0.0.5: use any username, for future releases: use 'octoprint' as username. Default ssh port is 8022 (same as termux)

Manaswin07, 2021-06-16:
Hi! Thanks for the quick reply. This works well.

closed 2021-06-16 by Manaswin07

MrDunham, 2022-04-03:
I'm reopening/commenting on this for the future semi-terminal literate people (like me) who are struggling with this.

To SSH in, open a terminal app (such as Terminal, on Mac)

Type:

ssh octoprint@localhost.local -p 8022

Enter the password you set when you turned on the SSH option on your Octo4a app (you did turn it on and set a password, right?)

Congratulations, you should now be in. Good luck and don't crater anything.

Londonbingbang, 2022-12-24:
[image]
How do i fix this?

mycream, 2023-02-06:
> [image] How do i fix this?
> 
> I need help setting Octo4a up (having problems with "installation corrupted" dm me on discord Redstonenee#9739 if you can help thanks)

Change "localhost.local" to the IP which you mobile phone WIFI used. Make sure your computer connect to the same WIFI.