feelfreelinux/octo4a #20: Octoprint is starting - never actually starts...

jaymcd, 2021-06-17:
Love this idea. My installation wont start though. It says it should only take a few seconds but it has been saying that for an hour now... Are there some additional permission or something I need to setup? Running on an old Asus Transformer TF300 running Katkiss Android ROM ( Android 7.1.2)

feelfreelinux, 2021-06-18:
This might be caused by a faulty install. Try to uninstall the app and install it again

jaymcd, 2021-06-18:
Ok. I will give it a try on Monday

Jay M. McDonald

jaymcd, 2021-06-21:
Works after reinstall. Thanks. Just working to get the Camera working. What are the normal Octoprint settings for it - anything different or special?

feelfreelinux, 2021-06-23:
> Works after reinstall. Thanks. Just working to get the Camera working. What are the normal Octoprint settings for it - anything different or special?

camera should be automatically configured in octo after the start. You just need to turn it on in my app's settings.

feelfreelinux, 2021-06-23:
Closing, as looks like the issue is resolved.

closed 2021-06-23 by feelfreelinux