feelfreelinux/octo4a #27: Build instructions

mmiscool, 2021-08-13:
Hello,
Love the project. 
Am interested in making a customized version with some octoprint plugins preinstalled and configured. 
Any guidance on build instructions and packaging would be appreciated.

feelfreelinux, 2021-08-13:
Hi! You should be able to achieve that by either editing the  installation script (https://github.com/feelfreelinux/octo4a/blob/master/app/app/src/main/java/com/octo4a/repository/OctoPrintHandlerRepository.kt#L93v) or by preparing a custom bootstrap linux rootfs. Hopefully I'll be able to implement some kind of init script customization with a new release.

feelfreelinux, 2021-09-29:
With 1.0.0 you can achieve that by modyfing the initial `setup-octo4a.py` script, or implementing an extension https://github.com/feelfreelinux/octo4a/wiki/Extensions-system

bzhmaddog, 2021-11-06:
I'd also like to be able to build the apk myself.

feelfreelinux, 2021-11-08:
After a latest commit, the main branch can be built by just importing the subfolder app into Android Studio.
Github workflow: https://github.com/feelfreelinux/octo4a/blob/master/.github/workflows/build-app.yml

As for the underlying bootstrap:
Scripts required to build it are available in this repository: https://github.com/feelfreelinux/android-linux-bootstrap
There's also a github action workflow that builds and releases it in that repository.

closed 2021-11-08 by feelfreelinux