feelfreelinux/octo4a #32: [Plugin] OctoEverywhere Webcam Snapshot and Stream Issues

QuinnDamerell, 2021-08-29:
Hello!

I'm the developer behind the OctoEverywhere service, a free service that empowers everyone with remote access to OctoPrint! A few of my users have been using the Android app setup and really enjoy it! However, there seem to be two issues, both relating to the webcam.

1. When my plugin tries to call the snapshot URL, the Android app crashes.
2. When my plugin tries to stream the webcam, the stream gets interrupted and closes quickly.

A bit of context on how my service works. The service is powered by a plugin users install in OctoPrint. The plugin keeps an active WebSocket connection to my service and my service relays HTTP requests from the user's browser to the plugin, the plugin makes the HTTP request locally, and then sends the result back.

For the snapshot issue, I have looked into it, but since the app is crashing I can't quite tell why. Are there any logs I can grab from the app that might give me some context?

About the webcam issue, I have been looking at your server logic and my PY logic. The PY logic for my plugin works for the default OctoPrint webcam server, but it's having trouble with your server for some reason. It looks like the request is closing right after sending just a bit of data. I'm still looking into this streaming issue and will update here if I find more!

feelfreelinux, 2021-08-30:
Thanks for letting me know! Since the lastest update (0.0.5) I've rewritten a lot of the camera server, so there's a chance that this bug is already fixed :) I'll try to reproduce it as soon as possible.

QuinnDamerell, 2021-08-31:
Great! Thanks for the update! 

Any idea when the new app version will be released? Or is it possible for me to get a test version of it to try out?

feelfreelinux, 2021-09-11:
Fixed in: https://github.com/feelfreelinux/octo4a/commit/cdd0e054471d9a83d4ca9afb7842afc7adf52bfe

You can either verify it by trying a build straight from artifacts, or wait a few days for 1.0.0 :) I'll keep the issue open until 1.0.0 release.

QuinnDamerell, 2021-09-11:
Great! Thanks so much!

Congrats on your 1.0 release! Once you release I will verify that the webcam works well and then would be happy to promote your work on my website!

QuinnDamerell, 2021-09-11:
Actually, along that line, can you shoot me a message over on my [support website](https://octoeverywhere.com/support) so we can discuss it more? :)

feelfreelinux, 2021-09-29:
1.0.0 was released a few days ago, would love to see you retest it :)
https://github.com/feelfreelinux/octo4a/releases/tag/1.0.0

QuinnDamerell, 2021-10-02:
Hey! 

Congrats on your app release! It's even better now and it still blows me away how easy it is to use! 

I did test the camera though and it's not working, at least on my Android phone. I don't have time now, but in a bit, I can keep testing and see if I can get more insight.