feelfreelinux/octo4a #34: Upload failed - Could not upload the file

ZeleiAttila, 2021-09-15:
Hi,

I've installed the app on a Minix U9-H (Android 7.1.2). When I upload a gcode I get a error message: Upload failed. I've granted storage permission in application settings, but I can see in log there is a permission error. 
Can you help me? 

Thanks

system info:
>browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/93.0.4577.63 Safari/537.36
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 8
env.hardware.freq: 1256
env.hardware.ram: 1755201536
env.os.bits: 64
env.os.id: linux
env.os.platform: linux
env.python.pip: 19.2.3
env.python.version: 3.8.0
env.python.virtualenv: false
octoprint.safe_mode: false
octoprint.version: 1.6.1
systeminfo.generator: systemapi

octoprint.log:

> 2021-09-14 22:21:02,430 - octoprint.startup - INFO - ******************************************************************************
2021-09-14 22:21:02,434 - octoprint.startup - INFO - Starting OctoPrint 1.6.1
2021-09-14 22:21:02,436 - octoprint.startup - INFO - ******************************************************************************
2021-09-14 22:21:02,462 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2021-09-14 22:21:02,463 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2021-09-14 22:21:02,463 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2021-09-14 22:21:02,565 - octoprint.plugin.core - INFO - Loading plugins from /data/data/com.octo4a/files/usr/lib/python3.8/site-packages/octoprint/plugins, /data/data/com.octo4a/files/home/.octoprint/plugins and installed plugin packages...
2021-09-14 22:21:06,072 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin (2021.8.2) did not pass check, not loading.
2021-09-14 22:21:06,376 - octoprint.plugin.core - INFO - Found 16 plugin(s) providing 16 mixin implementations, 32 hook handlers
2021-09-14 22:21:06,536 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2021-09-14 22:21:07,393 - octoprint.server - INFO - Intermediary server started
2021-09-14 22:21:07,402 - octoprint.printer.profile - ERROR - Profile _default does not exist, creating _default again and setting it as default
2021-09-14 22:21:07,633 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /storage/emulated/0/OctoPrint/uploads...
2021-09-14 22:21:07,637 - octoprint.filemanager.storage - INFO - ... file metadata for /storage/emulated/0/OctoPrint/uploads initialized successfully.
2021-09-14 22:21:07,645 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
2021-09-14 22:21:07,646 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
2021-09-14 22:21:07,647 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
2021-09-14 22:21:07,648 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
2021-09-14 22:21:07,650 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_manage'), Need(method='role', value='plugin_announcements_read')")
2021-09-14 22:21:07,652 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
2021-09-14 22:21:07,655 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')")
2021-09-14 22:21:07,658 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')")
2021-09-14 22:21:07,660 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')")
2021-09-14 22:21:07,663 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
2021-09-14 22:21:07,665 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: "Need(method='role', value='plugin_pluginmanager_install'), Need(method='role', value='plugin_pluginmanager_manage')")
2021-09-14 22:21:07,668 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: "Need(method='role', value='plugin_softwareupdate_check')")
2021-09-14 22:21:07,671 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: "Need(method='role', value='plugin_softwareupdate_update')")
2021-09-14 22:21:07,673 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CONFIGURE (needs: "Need(method='role', value='plugin_softwareupdate_configure')")
2021-09-14 22:21:10,907 - octoprint.util.pip - INFO - Using "/data/data/com.octo4a/files/usr/bin/python3 -m pip" as command to invoke pip
2021-09-14 22:21:14,021 - octoprint.util.pip - INFO - Version of pip is 19.2.3
2021-09-14 22:21:14,025 - octoprint.util.pip - INFO - pip installs to /data/data/com.octo4a/files/usr/lib/python3.8/site-packages (writable -> yes), --user flag needed -> no, virtual env -> no
2021-09-14 22:21:14,026 - octoprint.util.pip - INFO - ==> pip ok -> yes
2021-09-14 22:21:14,242 - octoprint.plugin.core - INFO - Initialized 16 plugin implementation(s)
2021-09-14 22:21:14,683 - octoprint.plugin.core - INFO - 16 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /data/data/com.octo4a/files/usr/lib/python3.8/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /data/data/com.octo4a/files/usr/lib/python3.8/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /data/data/com.octo4a/files/usr/lib/python3.8/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /data/data/com.octo4a/files/usr/lib/python3.8/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /data/data/com.octo4a/files/usr/lib/python3.8/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /data/data/com.octo4a/files/usr/lib/python3.8/site-packages/octoprint/plugins/backup
[... 92 lines cut ...]
  File "/data/data/com.octo4a/files/usr/lib/python3.8/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/data/data/com.octo4a/files/usr/lib/python3.8/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/data/data/com.octo4a/files/usr/lib/python3.8/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/data/data/com.octo4a/files/usr/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 1550, in decorated_view
    return func(*args, **kwargs)
  File "/data/data/com.octo4a/files/usr/lib/python3.8/site-packages/octoprint/vendor/flask_principal.py", line 199, in _decorated
    rv = f(*args, **kw)
  File "/data/data/com.octo4a/files/usr/lib/python3.8/site-packages/octoprint/server/api/files.py", line 641, in uploadGcodeFile
    added_file = fileManager.add_file(
  File "/data/data/com.octo4a/files/usr/lib/python3.8/site-packages/octoprint/filemanager/__init__.py", line 737, in add_file
    path_in_storage = self._storage(destination).add_file(
  File "/data/data/com.octo4a/files/usr/lib/python3.8/site-packages/octoprint/filemanager/storage.py", line 906, in add_file
    file_object.save(file_path)
[... cut ...]  in save
    shutil.move(self.path, path)
  File "/data/data/com.octo4a/files/usr/lib/python3.8/shutil.py", line 794, in move
    copy_function(src, real_dst)
  File "/data/data/com.octo4a/files/usr/lib/python3.8/shutil.py", line 427, in copy2
    copystat(src, dst, follow_symlinks=follow_symlinks)
  File "/data/data/com.octo4a/files/usr/lib/python3.8/shutil.py", line 366, in copystat
    lookup("utime")(dst, ns=(st.st_atime_ns, st.st_mtime_ns),
PermissionError: [Errno 1] Operation not permitted

th3Shah, 2021-09-23:
I have the same error but I upload to sd card and then refresh the sd card and the file appears. Mabe after an upload a wait time and then auto-refreshing the storage might help.

ZeleiAttila, 2021-09-28:
Hi,

I try the new release, but I got error when install, then I see "Installation corrupted" message in application.

Log:
💻 No bootstrap detected, proceeding with installation
💻 Downloading bootstrap 0.1.6
💻 Bootstrap extracted, setting it up...
🐸 add-user.sh
🐸 fake_proc_stat
🐸 install-bootstrap.sh
🐸 ioctlHook.so
🐸 root
🐸 rootfs.tar.xz
🐸 run-bootstrap.sh
🐸 install-bootstrap.sh[6]: ../root/bin/minitar: can't execute: Permission denied
🐸 install-bootstrap.sh[11]: can't create bootstrap/etc/resolv.conf: No such file or directory
🐸 bootstrap ready, run with run-bootstrap.sh
🐸 add-user.sh[5]: can't create bootstrap/etc/passwd: No such file or directory
🐸 run-bootstrap.sh[36]: ./root/bin/proot: can't execute: Permission denied
🐸 run-bootstrap.sh[36]: ./root/bin/proot: can't execute: Permission denied
🐸 run-bootstrap.sh[36]: ./root/bin/proot: can't execute: Permission denied
🐸 run-bootstrap.sh[36]: ./root/bin/proot: can't execute: Permission denied
💻 Bootstrap installation done
💻 Bootstrap installed
🐸 run-bootstrap.sh[36]: ./root/bin/proot: can't execute: Permission denied
🐸 run-bootstrap.sh[36]: ./root/bin/proot: can't execute: Permission denied
🐸 run-bootstrap.sh[36]: ./root/bin/proot: can't execute: Permission denied
🐸 run-bootstrap.sh[36]: ./root/bin/proot: can't execute: Permission denied
💻 Dependencies installed

feelfreelinux, 2021-09-28:
> Hi,
>
> I try the new release, but I got error when install, then I see "Installation corrupted" message in application.
>
> Log: 💻 No bootstrap detected, proceeding with installation 💻 Downloading [...]

Please try to do a fresh install, it should work now :) 

(Fixed in https://github.com/feelfreelinux/android-linux-bootstrap 0.1.8)

ZeleiAttila, 2021-09-29:
Hi,

Thank you. Everything works fine. The original issue was solved too.

closed 2021-09-29 by ZeleiAttila