feelfreelinux/octo4a #39: Octoprint doesn't restart instead keeps bug looping restart and crashing

th3Shah, 2021-09-30:
What I did to make my octoprint work:
- I made octoprint release 1.0.0 work on my android 7.1.1 (Asus Zenfone 2 Laser 2g ram) as well as an abandoned amazon firestick (gen2). I restart my device manually and then plug in the OTG cable (modified to not have the power go to the 3d printer) this prevents my phone from crashing phone to a forced power-off state. Then I did the initial setup and the server worked. As long as I don't restart the server for any reason it will not bug out or cause any other problems. The webserver works and I'm able to use octoprint with no other issue. 

What the issue is:
- Restarting octoprint doesn't restart at all or when it does it goes in the loop of shutting down and turning on. Then the ram gets hogged up and the app can crash mostly and sometimes freezes for a while. I prevent further freezing and crashing of the app by going to app storage in settings and clearing the data and reinstalling octoprint or doing a clean install. This prevents me from installing plugins nor am I able to go into safe mode.
(FYI if your device goes to complete black screen and the android doesn't turn on immediately give it 5 to 10 mins and keep it charging then just hold on to the power button and it should turn on in a minute or two)

Bellow are the data I was able to recover after replicating the issue. 
> octoprint.log :
2021-09-29 03:03:09,352 - octoprint.startup - INFO - ******************************************************************************
2021-09-29 03:03:09,356 - octoprint.startup - INFO - Starting OctoPrint 1.6.1
2021-09-29 03:03:09,362 - octoprint.startup - INFO - ******************************************************************************
2021-09-29 03:03:09,508 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2021-09-29 03:03:09,509 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2021-09-29 03:03:09,510 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2021-09-29 03:03:09,739 - octoprint.plugin.core - INFO - Loading plugins from /usr/lib/python3.9/site-packages/octoprint/plugins, /root/.octoprint/plugins and installed plugin packages...
2021-09-29 03:03:15,498 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin (2021.8.2) did not pass check, not loading.
2021-09-29 03:03:16,109 - octoprint.plugin.core - INFO - Found 16 plugin(s) providing 16 mixin implementations, 32 hook handlers
2021-09-29 03:03:16,435 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2021-09-29 03:03:16,631 - octoprint.server - INFO - Intermediary server started
2021-09-29 03:03:16,642 - octoprint.printer.profile - ERROR - Profile _default does not exist, creating _default again and setting it as default
2021-09-29 03:03:16,959 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /root/.octoprint/uploads...
2021-09-29 03:03:16,973 - octoprint.filemanager.storage - INFO - ... file metadata for /root/.octoprint/uploads initialized successfully.
2021-09-29 03:03:16,986 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
2021-09-29 03:03:16,988 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
2021-09-29 03:03:16,989 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
2021-09-29 03:03:16,990 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
2021-09-29 03:03:16,991 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_read'), Need(method='role', value='plugin_announcements_manage')")
2021-09-29 03:03:16,992 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
2021-09-29 03:03:16,995 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')")
2021-09-29 03:03:16,998 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')")
2021-09-29 03:03:17,002 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')")
2021-09-29 03:03:17,009 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
2021-09-29 03:03:17,010 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: "Need(method='role', value='plugin_pluginmanager_install'), Need(method='role', value='plugin_pluginmanager_manage')")
2021-09-29 03:03:17,011 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: "Need(method='role', value='plugin_softwareupdate_check')")
2021-09-29 03:03:17,012 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: "Need(method='role', value='plugin_softwareupdate_update')")
2021-09-29 03:03:17,013 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CONFIGURE (needs: "Need(method='role', value='plugin_softwareupdate_configure')")
2021-09-29 03:03:22,234 - octoprint.util.pip - INFO - Using "/usr/bin/python3 -m pip" as command to invoke pip
2021-09-29 03:03:27,082 - octoprint.util.pip - INFO - Version of pip is 20.3.4
2021-09-29 03:03:27,086 - octoprint.util.pip - INFO - pip installs to /usr/lib/python3.9/site-packages (writable -> yes), --user flag needed -> no, virtual env -> no
2021-09-29 03:03:27,087 - octoprint.util.pip - INFO - ==> pip ok -> yes
2021-09-29 03:03:27,487 - octoprint.plugin.core - INFO - Initialized 16 plugin implementation(s)
2021-09-29 03:03:28,269 - octoprint.plugin.core - INFO - 16 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/errortracking
|  File Check (2021.2.23) (bundled) = /usr/lib/python3.9/site-packages/octoprint_file_check
|  Firmware Check (2021.8.11) (bundled) = /usr/lib/python3.9/site-packages/octoprint_firmware_check
|  GCode Viewer (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/gcodeviewer
|  Logging (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/logging
|  Plugin Manager (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/virtual_printer
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
2021-09-29 03:03:28,414 - octoprint.environment - INFO - Detected environment is Python 3.9.5 under Linux (linux). Details:
|  hardware:
|    cores: 8
|    freq: 1247.6999999999998
|    ram: 1944424448
|  os:
|    bits: 64
[... cut ...]
💻 Failed to start. OctoPrint already running.
💻 Got event {"eventType": "restartServer"}
💻 Got event {"eventType": "restartServer"}
🐸 sh: mkfifo: not found
💻 Failed to start. OctoPrint already running.
💻 Failed to start. OctoPrint already running.
💻 Failed to start. OctoPrint already running.
💻 Failed to start. OctoPrint already running.
💻 Failed to start. OctoPrint already running.
💻 Failed to start. OctoPrint already running.
💻 Got event {"eventType": "restartServer"}
💻 Got event {"eventType": "restartServer"}

feelfreelinux, 2021-09-30:
Thanks for your detailed description. I've narrowed down the issue to missing "mkfifo" on some devices. Will fix it asap!

feelfreelinux, 2021-09-30:
Released 1.0.1 that fixes this issue.
https://github.com/feelfreelinux/octo4a/commit/10c35d0f348d7157cade040b237919665677cad8

closed 2021-09-30 by feelfreelinux