feelfreelinux/octo4a #40: proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory

tobyworks, 2021-10-03:
Whenever running a command that requires root im getting the above error and the command fails. 

`chmod +x` for instance

feelfreelinux, 2021-10-03:
What Android version are you running octo4a on?

tobyworks, 2021-10-03:
I'm running the source code from android studio arctic fox. I needed to
downgrade Gradle plugin to 7.0.1.

Compiling app to OnePlus Nord n100 running android 11.

Just to note the octoprint server is running fine. But I just notice that
running executable binary files is not possible

feelfreelinux, 2021-10-03:
> I'm running the source code from android studio arctic fox. I needed to downgrade Gradle plugin to 7.0.1. Compiling app to OnePlus Nord n100 running android 11. Just to note the octoprint server is running fine. But I just notice that running executable binary files is not possible
> […](#)

this is because of bindings inside octo4a. Everything under /root is bind to ~/storage/emulated/0/Android/data/com.octo4a. Internal storage in Android does not allow for execution of binary files, that's why in order to run anything inside of the bootstrap, you need to run it from any other path (like /whatever).

If you're hacking with octo4a let me know, there's quite a lot of weird hacks going on with the bootstrap.

tobyworks, 2021-10-03:
I actually am hacking with octo4a. Wanna sync up in discord ?

feelfreelinux, 2021-10-03:
> I actually am hacking with octo4a. Wanna sync up in discord ?

Sure, my tag is in my github bio :)

closed 2021-10-03 by feelfreelinux

Celiiiiiii, 2022-12-02:
so what was the solution ?

sensaurabh, 2023-03-16:
Looking for the solution too! Any help?

Angelevskioctoprint, 2023-06-04:
Same problem here, any help?