feelfreelinux/octo4a #54: Several Camera Issues

Nazar78, 2021-11-04:
Hi,

The latest update did correct the green screen. However I've noticed several other issues using the camera server.

1. After first few once or twice autofocus done perfectly, both the video and snapshots will eventually turn out of focus forever. It's that bad that I can't even see the BL Touch clearly. I've ensured the option is set not to disabled auto focus. Have toggle it still no change.

2. It's working in the previous version but now the flash seems to work only with video. During time lapse snapshot i.e. every 10 secs there's no flash, thus picture turns out dark, worst during the night. If I continue viewing the mjpeg stream during time lapse, only then the LED lights up. I'm using the flash in an enclosure.

3. My default setup rear camera seems to be detected as much more limited than the front facing. I can get 1920x1080p with the front but the rear max is only 2048x1536p. I'm using a rooted Note 3 LineageOS 18.1. Since with all these issues, I'm using the IP Webcam app instead and I can use 4K (overkill, I'd stay with 1080p, heat/battery wise).

I would much prefer using the camera with Octo4a, which seems to be using lesser resources shown in htop but unfortunately the camera is unusable now.  BTW any pointer how can we debug it with Android Studio? I fire it up thinking to look for answers but got lost at setting up the dependencies, little experience in java/kotlin.

Thank you.

feelfreelinux, 2021-11-08:
Hi, since one of the newest commits, the entire project should be easily importable into Android Studio as I moved the build to make. I'll try to address those camera issues soon, but if you want to try to debug it yourself, all of camera logic is in the https://github.com/feelfreelinux/octo4a/tree/master/app/app/src/main/java/com/octo4a/camera subfolder, mainly `CameraSevice.kt`. Thanks for reporting the issues!

Woytacze, 2021-11-12:
I am testing 1.0.4 on Lenovo Pad TB3-710F with stock android 5.0.1 (TB3-710F_170316).
1) After instalation I had tablet in counter clockwise landscape position and when I enable camera server app crashed and was crashing immediately at launch and I had to remove all and do instal over again.
2) According green screen. For booth cameras preview looks OK but on PC i see black and white video with green filter over it.