feelfreelinux/octo4a #55: Connection to printer failed

radimsejk, 2021-11-05:
I have instaled the app and connected to my printer (I3 MK3S). My phone showed 

> Printer is connected /dev/bus/usb/001/003
Original Prusa I3 MK3
VID2c99 / PID 2
CDCserial driver

but on web UI I cannot connect (everything is set to AUTO) and the state is

> State: Offline after error
No more candidates to test, and no working port/baudrate combination detected.

I have LG ThinQ V40 and Android 10.

Could you please help me with this? Should I add some additional info? 
Thank you.

feelfreelinux, 2021-11-05:
Can you try to force kill the app and try to connect again? I noticed a problem like that last time.

radimsejk, 2021-11-05:
I could not kill the app (I cannot see it in opened apps) so I have tried to reboot the phone. Then it rebooted the printer but still no connection. I tried to reboot the octoprint from the web UI and now it is finally connected. I hope it will work because running octoprint on the phone is awesome idea 👍

feelfreelinux, 2021-11-05:
Fixed in: https://github.com/feelfreelinux/octo4a/commit/34416c6aa792a662a41cb67b1b1afd038a4c70e3

Fix released with 1.0.4

closed 2021-11-05 by feelfreelinux