feelfreelinux/octo4a #56: Permission denied: '/dev/ttyACM0' (Galaxy A3 2017 + Monoprice Mini Select v2)

demianovics, 2021-11-05:
Terminal log of "Connect" on Serial Port "/dev/ttyACM0" on Baudrate "AUTO":
```
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: /dev/ttyACM0@115200, /dev/ttyACM0@250000, /dev/ttyACM0@230400, /dev/ttyACM0@57600, /dev/ttyACM0@38400, /dev/ttyACM0@19200, /dev/ttyACM0@9600
Trying port /dev/ttyACM0, baudrate 115200
Connecting to: /dev/ttyACM0
Using Malyan/Monoprice Connection Fix Plugin to create serial connection
Unexpected error while connecting to serial port /dev/ttyACM0, baudrate 115200 from hook malyan_connection_fix: SerialException: '[Errno 13] could not open port /dev/ttyACM0: [Errno 13] Permission denied: '/dev/ttyACM0'' @ comm.py:_open_serial:3775
Could not open port /dev/ttyACM0, baudrate 115200, skipping
...
Changing monitoring state from "Detecting serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
```

Switching through the usb-drivers in Octo4a doesn't make a difference, except that some won't show "/dev/ttyACM0" as "Serial Port" option.

Also, un- and re-plugging the usb-cable (no matter if i change drivers in between), always makes Android pop up a dialog asking me to grant Octo4a the right to access the usb-device. Clicking the "Always allow" checkbox has no effect, and the dialog always reappears.

Any idea?

feelfreelinux, 2021-11-05:
Hi! What printer model are you connecting it to?  Does octo4a automatically detect the driver? Please select /dev/ttyOcto4a as it's 100% the correct option.

demianovics, 2021-11-06:
Hi! It's a Monoprice Select Mini v2. My phone is a Samsung Galaxy A3 (2017).
I tried again with /dev/ttypOtco4a and all Baudrates, each manually and AUTO.

```
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to: /dev/ttyOcto4a
Using Malyan/Monoprice Connection Fix Plugin to create serial connection
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0xf04b0ee0, open=True>(port='/dev/ttyOcto4a', baudrate=38400, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
No answer from the printer within the connection timeout, trying another hello
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor
```

When connecting the printer via usb, it detects and shows it as
```
Malayan 3D Printer
VID 2e26 / PID 200
FDTIserial driver(tap to select)
```

feelfreelinux, 2021-11-08:
Hi! Hope you had a good weekend. The printer was not detected successfully, and looks like the selected driver is wrong. Can you please click on the (tap to change) field and change it to CH340? After that `AUTO` with `/dev/ttyOcto4a` should work fine :)

demianovics, 2021-11-08:
Hi, i had a great weekend :) Hope you too. I tried again, with driver "CH341" (CH340 isn't listed). Then "AUTO" baudrate. But again, no luck :(

```
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: /dev/ttyOcto4a@115200, /dev/ttyOcto4a@250000, /dev/ttyOcto4a@230400, /dev/ttyOcto4a@57600, /dev/ttyOcto4a@38400, /dev/ttyOcto4a@19200, /dev/ttyOcto4a@9600
Trying port /dev/ttyOcto4a, baudrate 115200
Connecting to: /dev/ttyOcto4a
Using Malyan/Monoprice Connection Fix Plugin to create serial connection
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0xf3159ec8, open=True>(port='/dev/ttyOcto4a', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyOcto4a, baudrate 250000
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyOcto4a, baudrate 230400
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyOcto4a, baudrate 57600
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyOcto4a, baudrate 38400
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyOcto4a, baudrate 19200
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyOcto4a, baudrate 9600
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Changing monitoring state from "Detecting serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
```

Am i right assuming, that Octo4a is a wrapper for Octoprint? So if i don't manage to get that printer connected to Octoprint on my Raspberry Pi 3B+, the problem has nothing to do with Otco4a? Maybe i find the time to give that rabbithole a try.

Still, i'd be very interested in any suggestion you have. Thank you very much so far. Dzięki.

feelfreelinux, 2021-11-08:
Maybe I wasn't right with the ch34x selection, please try the two other choices (pl and CDC) too

demianovics, 2021-11-08:
I tried all of them drivers. With different cables. Out of hundreds of tries, there was one time, where after `Send: N0 M110 N0*125` two immediate replies followed. It looked like it was connected. I think it was the CDC driver. But the connection broke down shortly afterwards.
I think it was the CDC driver, /dev/ttyOtco4a and 115200 baudrate, Monoprice plugin disabled.

But, till that point, i was switching through the drivers maniacally. Also plugging-off and -on the cable every now and then. And sometimes restarting the 3d printer or Octo4a (the server) or disabling the. I think a have tried 200 times in different combinations. And i couldn't recreate that one time when it worked :( I am going nuts.

Can you tell me, if switching the driver has an immediate effect? Or do i have to un- and replug the usb?

feelfreelinux, 2021-11-08:
Replugging the cable is probably necessary

feelfreelinux, 2021-11-08:
Looks like monoprice needs a custom open sequence, will look into it

demianovics, 2021-11-09:
Just so that you know, i have googled and it seems that others have achieved connecting to that printer with Octoprint. They report that they need 11520 baudrate. And that they need to click connect and disconnect a couple of times. But more often than not that seems to work. Then that Monoprice plugin for Octcoprint appeared, and it seems like it fixed the problem for most of them.

So i'd say that in general, Monoprice and Octoprint work well.

Whats special in the Otco4a case, i can't tell. Maybe it's my setup only.

Before trying it with a Pi 3B+, i might try it with my Huawei Media Pad M5 lite and Octo4a tomorrow.

Thanks for your efforts!

miserovitch, 2021-11-12:
Hi guys,

For info I have exactly the same issue when trying to connect my honor 8 and a BTT GT V1.0 motherboard.
Hope you gonna come with great ideas to solve it!

Thank you

SoM

D5KDeutsche, 2023-01-06:
Having the same issue. I can connect just fine using Repetier Host on Windows, but not Octo4a.

alufers, 2023-03-21:
Try one of the newer versions.

closed 2023-03-21 by alufers