feelfreelinux/octo4a #68: Installation stage fails and app crashes

rahultadak, 2021-11-08:
Hi,

I am trying to install octo4a on a Google Pixel XL. First I tried installing the 1.0.4 version which kept crashing before completing the install.

I then tried with the 1.0.3 version. This time the installation didn't crash, but definitely failed. At least this time the app didn't crash so I could get the logs.

💻 No bootstrap detected, proceeding with installation
💻 Downloading bootstrap 0.1.8
💻 Bootstrap extracted, setting it up...
🐸 add-user.sh
🐸 fake_proc_stat
🐸 install-bootstrap.sh
🐸 ioctlHook.so
🐸 root
🐸 rootfs.tar.xz
🐸 run-bootstrap.sh
🐸 bootstrap ready, run with run-bootstrap.sh
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 Welcome to Alpine!
🐸 
🐸 The Alpine Wiki contains a large amount of how-to guides and general
🐸 information about administrating Alpine systems.
🐸 See <http://wiki.alpinelinux.org/>.
🐸 
🐸 You can setup the system with the command: setup-alpine
🐸 
🐸 You may change this message by editing /etc/motd.
🐸 
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 fetch https://dl-cdn.alpinelinux.org/alpine/v3.14/main/aarch64/APKINDEX.tar.gz
🐸 ERROR: https://dl-cdn.alpinelinux.org/alpine/v3.14/main: temporary error (try again later)
🐸 WARNING: Ignoring https://dl-cdn.alpinelinux.org/alpine/v3.14/main: No such file or directory
🐸 fetch https://dl-cdn.alpinelinux.org/alpine/v3.14/community/aarch64/APKINDEX.tar.gz
🐸 ERROR: https://dl-cdn.alpinelinux.org/alpine/v3.14/community: temporary error (try again later)
🐸 WARNING: Ignoring https://dl-cdn.alpinelinux.org/alpine/v3.14/community: No such file or directory
🐸 ERROR: unable to select packages:
🐸   bash (no such package):
🐸     required by: world[bash]
🐸   curl (no such package):
🐸     required by: world[curl]
🐸   openssh-server (no such package):
🐸     required by: world[openssh-server]
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 /bin/sh: can't create /etc/ssh/sshd_config: nonexistent directory
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 /bin/sh: ssh-keygen: not found
💻 Bootstrap installation done
💻 Bootstrap installed
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 /bin/sh: curl: not found
🐸 /bin/sh: bash: not found
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 /bin/sh: curl: not found
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 unzip: can't open octoprint.zip[.zip]
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 /bin/sh: cd: line 1: can't cd to Octo*: No such file or directory
💻 Dependencies installed

Not sure what I'm missing. I'm on the last security patch for Pixel XL from October 2019.

feelfreelinux, 2021-11-08:
Hi! The error is caused by your phone not being able to connect to alpine mirrors. By any chance are you in China / behind a firewall?

rahultadak, 2021-11-08:
I'm in the Netherlands. I have a pi.hole running, which I did pause and try installing. That didn't help either.

feelfreelinux, 2021-11-08:
> I'm in the Netherlands. I have a pi.hole running, which I did pause and try installing. That didn't help either.

Are you able to access https://dl-cdn.alpinelinux.org/alpine/v3.14/main/aarch64/APKINDEX.tar.gz ?

rahultadak, 2021-11-08:
Yeah I was able download that file.

rahultadak, 2021-11-08:
The proot command seems to fail before fetching from Alpine. Do you think that has something to do witb it?

feelfreelinux, 2021-11-08:
> The proot command seems to fail before fetching from Alpine. Do you think that has something to do with it?

the proot command is just a warning and it's supposed to fail this way at that stage. It's purely the fault of octo4a not being able to reach the alpine servers but I'm not sure why it happens if you can download that URL from your phone. If you have access to any VPN software, using it might change something.

rahultadak, 2021-11-08:
Wow. I used some basic free VPN and it seems to be installing now. I'll update once it completes.

rahultadak, 2021-11-08:
That worked. I'm able to setup octoprint now!

closed 2021-11-08 by rahultadak

royaltongue, 2023-05-18:
@rahultadak What VPN did you use? I tried NordVPN and it still failed