feelfreelinux/octo4a #70: Issue connecting to Creality SE6

SirDragonClaw, 2021-11-10:
Hey so I have been trying to get this working on my Galaxy s9+ and I have the camera working perfectly now, but when I actually try to connect to the printer I get an error.

The error seems to indicate a permissions issue, I made sure to give permission to the serial port when I first plugged in the printer. I have used this printer on a raspberry pi 2 in the past and it worked fine, so I don't think its that.

I have tried a few high quality cables, and still get the same error. 

Is there some additional permission I need to give this application in android in order for it to have reliable serial access?

Thanks!

`Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 2 port/baudrate candidates: /dev/ttySAC1@115200, /dev/ttySAC0@115200
Trying port /dev/ttySAC1, baudrate 115200
Connecting to port /dev/ttySAC1, baudrate 115200
Unexpected error while connecting to serial port /dev/ttySAC1, baudrate 115200 from hook default: SerialException: '[Errno 13] could not open port /dev/ttySAC1: [Errno 13] Permission denied: '/dev/ttySAC1'' @ comm.py:_open_serial:3775
Could not open port /dev/ttySAC1, baudrate 115200, skipping
Trying port /dev/ttySAC0, baudrate 115200
Connecting to port /dev/ttySAC0, baudrate 115200
Unexpected error while connecting to serial port /dev/ttySAC0, baudrate 115200 from hook default: SerialException: '[Errno 13] could not open port /dev/ttySAC0: [Errno 13] Permission denied: '/dev/ttySAC0'' @ comm.py:_open_serial:3775
Could not open port /dev/ttySAC0, baudrate 115200, skipping
Changing monitoring state from "Detecting serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"`

feelfreelinux, 2021-11-10:
Hi! By any chance, have you restored a backup? the selected serial port needs to be set to `/dev/ttyOcto4a`, picking anything else won't work. In case you updated from a backup, then to get `/dev/ttyOcto4a` as an option, add it to additional serial ports list in octoprint's settings.

SirDragonClaw, 2021-11-11:
Why yes I did. And adding that manually did fix it. 
I tried to restore from a backup to make things easier but all it did was break the webcam and break the serial haha. Maybe add a suggestion in the readme to not restore from a backup or how to fix it if you do.

Thanks for the fast reply!

closed 2021-11-11 by SirDragonClaw