feelfreelinux/octo4a #72: Rear Facing camera option

AdamsLab01, 2021-11-11:
Hello, 

I'm running octo4a on a LG G7 ThinQ G710U which has two rear cameras (normal and wide). While octo4a gives me two rear camera options, selecting one rear camera over the other does not switch between the cameras, it just stays on the normal camera. Not a big deal, just an FYI as it would be nice to use the wide camera. 

Also, I've noticed that the camera resolution setting does not survive a restart, it always defaults to a lower resolution than what I set it to.

feelfreelinux, 2021-11-12:
Thanks! Will improve it soon

richyr45, 2022-10-16:
Hey! Any news on getting an secondary rear cam to work?
With my LG G6 (Front + main read + wide rear) I still only can choose between front and main rear – as stated by Adam there is a second read choise but doesnt change anything.
Im on 1.8.4.
Thanks for your great work, Richy