feelfreelinux/octo4a #77: Installation corrupt - something about serialpipe?

frankkienl, 2021-11-12:
I tried to install octo4a on a Sony phone, running Android 5.0.2
Specs: https://www.gsmarena.com/sony_xperia_z2-6144.php
And ran into a problem with the installation, it says installation corrupt.
Retried the install, reinstalled the app, rebooted the phone. Didn't work.
The log seems to suggest a missing file?
 
```
2021-11-12 16:33:46.101 13313-13395/? V/TAG: PtyThread getting ready
2021-11-12 16:33:46.103 13313-13395/? V/TAG: PtyThread called openpty
2021-11-12 16:33:46.103 13313-13395/? V/TAG: SYMLIKED
2021-11-12 16:33:46.145 13313-13313/? V/GetEnv::  Subscribe to Listener  OK 
2021-11-12 16:33:46.697 13313-13313/? I/Timeline: Timeline: Activity_idle id: android.os.BinderProxy@3c7eb85a time:305081
2021-11-12 16:33:46.994 13313-13396/? D/��: No bootstrap detected, proceeding with installation
2021-11-12 16:33:47.720 13313-13397/? D/��: Downloading bootstrap 0.1.8
2021-11-12 16:33:49.142 13313-13313/? W/BugsnagNDK: Initialization complete!
2021-11-12 16:33:49.166 13313-13349/? D/OpenGLRenderer: endAllStagingAnimators on 0x9b92eb00 (RippleDrawable) with handle 0x9de32790
2021-11-12 16:33:50.437 13313-13397/? D/��: Bootstrap extracted, setting it up...
2021-11-12 16:33:50.479 13313-13397/? D/��: add-user.sh
2021-11-12 16:33:50.479 13313-13397/? D/��: fake_proc_stat
2021-11-12 16:33:50.479 13313-13397/? D/��: install-bootstrap.sh
2021-11-12 16:33:50.480 13313-13397/? D/��: ioctlHook.so
2021-11-12 16:33:50.480 13313-13397/? D/��: root
2021-11-12 16:33:50.480 13313-13397/? D/��: root-pre5
2021-11-12 16:33:50.481 13313-13397/? D/��: rootfs.tar.xz
2021-11-12 16:33:50.482 13313-13397/? D/��: run-bootstrap.sh
2021-11-12 16:33:50.557 13313-13397/? D/��: Bad mode
2021-11-12 16:33:50.730 13313-13397/? D/��: bootstrap ready, run with run-bootstrap.sh
2021-11-12 16:33:50.847 13313-13397/? D/��: proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
2021-11-12 16:33:50.866 13313-13397/? D/��: Error loading shared library libcutils.so: No such file or directory (needed by /system/lib/stamina.so)
2021-11-12 16:33:50.867 13313-13397/? D/��: Error loading shared library libstdc++.so: No such file or directory (needed by /system/lib/stamina.so)
2021-11-12 16:33:50.867 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __android_log_assert: symbol not found
2021-11-12 16:33:50.868 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __strlen_chk: symbol not found
2021-11-12 16:33:50.868 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __errno: symbol not found
2021-11-12 16:33:50.921 13313-13397/? D/��: proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
2021-11-12 16:33:50.937 13313-13397/? D/��: Error loading shared library libcutils.so: No such file or directory (needed by /system/lib/stamina.so)
2021-11-12 16:33:50.939 13313-13397/? D/��: Error loading shared library libstdc++.so: No such file or directory (needed by /system/lib/stamina.so)
2021-11-12 16:33:50.940 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __android_log_assert: symbol not found
2021-11-12 16:33:50.941 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __strlen_chk: symbol not found
2021-11-12 16:33:50.942 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __errno: symbol not found
2021-11-12 16:33:50.986 13313-13397/? D/��: proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
2021-11-12 16:33:50.998 13313-13397/? D/��: Error loading shared library libcutils.so: No such file or directory (needed by /system/lib/stamina.so)
2021-11-12 16:33:51.000 13313-13397/? D/��: Error loading shared library libstdc++.so: No such file or directory (needed by /system/lib/stamina.so)
2021-11-12 16:33:51.001 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __android_log_assert: symbol not found
2021-11-12 16:33:51.002 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __strlen_chk: symbol not found
2021-11-12 16:33:51.003 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __errno: symbol not found
2021-11-12 16:33:51.054 13313-13397/? D/��: proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
2021-11-12 16:33:51.071 13313-13397/? D/��: Error loading shared library libcutils.so: No such file or directory (needed by /system/lib/stamina.so)
2021-11-12 16:33:51.072 13313-13397/? D/��: Error loading shared library libstdc++.so: No such file or directory (needed by /system/lib/stamina.so)
2021-11-12 16:33:51.072 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __android_log_assert: symbol not found
2021-11-12 16:33:51.073 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __strlen_chk: symbol not found
2021-11-12 16:33:51.074 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __errno: symbol not found
2021-11-12 16:33:51.076 13313-13397/? D/��: Bootstrap installation done
2021-11-12 16:33:51.076 13313-13397/? D/��: Bootstrap installed
2021-11-12 16:33:51.085 13313-13313/? V/GetEnv::  Subscribe to Listener  OK 
2021-11-12 16:33:51.122 13313-13397/? D/��: proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
2021-11-12 16:33:51.135 13313-13397/? D/��: Error loading shared library libcutils.so: No such file or directory (needed by /system/lib/stamina.so)
2021-11-12 16:33:51.136 13313-13397/? D/��: Error loading shared library libstdc++.so: No such file or directory (needed by /system/lib/stamina.so)
2021-11-12 16:33:51.137 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __android_log_assert: symbol not found
2021-11-12 16:33:51.137 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __strlen_chk: symbol not found
2021-11-12 16:33:51.138 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __errno: symbol not found
2021-11-12 16:33:51.184 13313-13397/? D/��: proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
2021-11-12 16:33:51.196 13313-13397/? D/��: Error loading shared library libcutils.so: No such file or directory (needed by /system/lib/stamina.so)
2021-11-12 16:33:51.197 13313-13397/? D/��: Error loading shared library libstdc++.so: No such file or directory (needed by /system/lib/stamina.so)
2021-11-12 16:33:51.198 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __android_log_assert: symbol not found
2021-11-12 16:33:51.199 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __strlen_chk: symbol not found
2021-11-12 16:33:51.199 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __errno: symbol not found
2021-11-12 16:33:51.258 13313-13397/? D/��: proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
2021-11-12 16:33:51.273 13313-13397/? D/��: Error loading shared library libcutils.so: No such file or directory (needed by /system/lib/stamina.so)
2021-11-12 16:33:51.274 13313-13397/? D/��: Error loading shared library libstdc++.so: No such file or directory (needed by /system/lib/stamina.so)
2021-11-12 16:33:51.275 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __android_log_assert: symbol not found
2021-11-12 16:33:51.275 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __strlen_chk: symbol not found
2021-11-12 16:33:51.276 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __errno: symbol not found
2021-11-12 16:33:51.285 13313-13313/? V/GetEnv::  Subscribe to Listener  OK 
2021-11-12 16:33:51.347 13313-13397/? D/��: proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
2021-11-12 16:33:51.366 13313-13397/? D/��: Error loading shared library libcutils.so: No such file or directory (needed by /system/lib/stamina.so)
2021-11-12 16:33:51.368 13313-13397/? D/��: Error loading shared library libstdc++.so: No such file or directory (needed by /system/lib/stamina.so)
2021-11-12 16:33:51.370 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __android_log_assert: symbol not found
2021-11-12 16:33:51.371 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __strlen_chk: symbol not found
2021-11-12 16:33:51.372 13313-13397/? D/��: Error relocating /system/lib/stamina.so: __errno: symbol not found
2021-11-12 16:33:51.375 13313-13313/? V/GetEnv::  Subscribe to Listener  OK 
```

jcpalumbo, 2021-11-22:
When you are installing you should get a notification to allow the installer to write to your phones storage. I totally missed this on my first try and had the same issues as you stated. The install will complete as fast once you give it the right permissions.

OzGav, 2022-04-11:
Yeah not sure if it is a permissions thing. I could only get 0.0.5 to install properly on my HTC One A9