feelfreelinux/octo4a #78: Installation ALWAYS says its corrupt

stevvie, 2021-11-13:
Tried to install on a samsung note 10.1 (2014 tablet) but the installation always goes buy real fast (about 5-10 seconds) and then disappears with an icon in the status bar saying octoprint server is starting up. After clicking the notification I get installation corrupted net to the green printer icon. I clicked the broken heart and I can try and install again but its a vicious circle. I orrigionally tried this on android 7.1 and then on android 10 after a factory reset and twrp a new rom/gapps. I can't even read the log as it goes too fast to see

jcpalumbo, 2021-11-16:
permissions issue

stevvie, 2021-11-17:
Thanks for the reply, but what is the answer ????

jcpalumbo, 2021-11-22:
When you are installing you should get a notification to allow the installer to write to your phones storage.  I totally missed this on my first try and had the same issues as you stated.   The install will complete as fast once you give it the right permissions.

stevvie, 2021-11-22:
Yes get the message and click allow. I've secsefully installed this on a Zuk Z2 running androis 7.12. But just unable to install on my samsung Note 10.1 even after clicking allow on Android 7.0 then android 7.1 and now even tried android 10 (LineageOS 17.1).

wjflier, 2022-06-09:
I have 8 phones, only 1 will  allow installation.