feelfreelinux/octo4a #81: Offline after error

norbdev, 2021-11-13:
**Error:**
State: Offline after error
No more candidates to test, and no working port/baudrate combination detected.

Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 0 port/baudrate candidates: 
Changing monitoring state from "Detecting serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"

Tried 3 phones, 4 OTG cables and the error message the same. If I unplug the phone the error message still the same.

Every phone:
Printer is connected
FT232R USB UART
VIN 403 / PID 6001
FTDI serial driver

printer: CR-10 with Marlin 2.0.7.2

norbdev, 2021-11-13:
apk version 0.0.5 working

norbdev, 2021-11-13:
apk version 0.0.5 working, it detects the port "/dev/ttyOcto4a"
if I add the port to the "Additional serial ports" it works with the latest apk

alufers, 2021-11-29:
Yes, please add `/dev/ttyOcto4a` to Additional serial ports in octoprint settings.

closed 2021-11-29 by alufers

brolly916, 2022-02-22:
my cr10 s5 is being connected to my android box  via a normal USB cable and the octo4a application is able to detact printer at /dec/bus/usb/001/002

tried to add /dec/bus/usb/001/002 Additional serial ports in octoprint settings to no avail

/dev/ttyOcto4a is already added to Additional serial ports in octoprint settings however from the octoprint dashboard this port is not showing up can someone help me please ?