feelfreelinux/octo4a #83: Crash on pressing extentions-tab

frankkienl, 2021-11-15:
App crashes when pressing on the extentions-tab; Version 1.0.4;
Device:  Motorola One
https://www.gsmarena.com/motorola_one_(p30_play)-9310.php

```
2021-11-15 18:02:25.725 14470-14470/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.octo4a, PID: 14470
    java.lang.NullPointerException: Attempt to get length of null array
        at com.octo4a.repository.ExtensionsRepositoryImpl.getValidExtensions(ExtensionsRepository.kt:213)
        at com.octo4a.ui.fragments.ExtensionsFragment.onViewCreated(ExtensionsFragment.kt:32)
        at androidx.fragment.app.Fragment.performViewCreated(Fragment.java:2974)
        at androidx.fragment.app.FragmentStateManager.createView(FragmentStateManager.java:543)
        at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:282)
        at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:2177)
        at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:2094)
        at androidx.fragment.app.FragmentManager.execPendingActions(FragmentManager.java:1990)
        at androidx.fragment.app.FragmentManager$5.run(FragmentManager.java:524)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.app.ActivityThread.main(ActivityThread.java:6912)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:860)
```

feelfreelinux, 2021-11-16:
Looks like a corrupt install, please reinstall the app

frankkienl, 2021-11-16:
Works after re-install. Thanks!

closed 2021-11-16 by frankkienl