feelfreelinux/octo4a #85: Successfully connected, web interface stuck on 'connecting'

realmadridsi, 2021-11-16:
I seem to have the problem that when I connect to my printer it successfully connects according to the terminal, but the web interface is stuck on 'connecting' and I'm not able to do anything but to disconnect again. I also do receive temperature information.

[image]

feelfreelinux, 2021-11-16:
Hm, the result from the printer seems to be formatted pretty weirdly. What firmware do you have on your ender?

realmadridsi, 2021-11-16:
> Hm, the result from the printer seems to be formatted pretty weirdly. What firmware do you have on your ender?

Running stock firmware. Creality doubles some information in their stock firmware. There is a plugin to fix this.