feelfreelinux/octo4a #86: Not sure where to find API Key

IZZYGamez361, 2021-11-18:
I have the OctoPrint plugin for Ucltimaker Cura 4.12.0. I have all set and ready but this API thing is stopping me.

impatman, 2021-11-20:
This is not related octo4a, this is for general Octoprint. https://all3dp.com/2/cura-octoprint-plugin-connection/

IZZYGamez361, 2021-11-20:
I tried getting the API Key but that page never loads.

IZZYGamez361, 2021-11-20:
> This is not related octo4a, this is for general Octoprint. https://all3dp.com/2/cura-octoprint-plugin-connection/

I tried using the IP given to find the API Key, but it just sits and buffers. Is that not a fault of octo4a or OctoPrint?

alufers, 2021-11-29:
Hey! This indeed looks like a problem with octoprint and not octo4a.

If it turns out that octoprint doesn't even start for you because of an octo4a issue, please open a new issue.

closed 2021-11-29 by alufers