feelfreelinux/octo4a #93: OTG host cable problem

mchenier, 2021-11-24:
This is not a software problem but I think it's the best place to ask this question.

I bought a OTG Host cable but the problem I encounter is that the cable is made so the charge goes to the accessory not the phone. So the 3d Printer gets the power witch is unecessary. 
Did anybody find a cable that charge the phone and not the accessory you plug in the cable?

zeatful, 2021-11-24:
Are you using the same type of cable / or the exact one as mentioned in the wiki: https://github.com/feelfreelinux/octo4a/wiki/Connecting-to-printer-via-OTG,-simultaneous-charging?

mchenier, 2021-11-25:
Yes I have exactly this cable. I guess my old Moto G Play don't support that. Thanks

andreasmo, 2021-12-04:
Although it does not apply to the Moto G Play I had a similar problem with the Moto 4 X. It seems only adapters which support the USB Power Delivery protocol can be used for simultaneous connection and charging. 
For me this one worked:
https://de.aliexpress.com/item/1005003324941009.html

wjflier, 2022-05-19:
I just soldered a charge cable to the two battery pins in the phone