feelfreelinux/octo4a #94: POCO M3

tanjunior, 2021-11-25:
Have anyone tried with a POCO M3?

I bought this hub but it doesn't support charging and data simultaneously. 
[https://www.aliexpress.com/item/1005001694740332.html](https://www.aliexpress.com/item/1005001694740332.html)

I am not sure if it is a ROM issue or Hub issue, would be great if anyone could share their experience with POCO M3.
So that I can buy a new Hub if it is a Hub issue.

stevvie, 2021-11-25:
I have tried 3 hubs/splitters and none charge and pass through OTG data at the same time on my Poco F3 or ZUK Z2Plus.

tanjunior, 2021-11-26:
> I have tried 3 hubs/splitters and none charge and pass through OTG data at the same time on my Poco F3 or ZUK Z2Plus. Almost certain it's a hub/cable issue as I've tried this on different roms.

Thank you for your feedback. If you are 100% sure it's a hub issue then I will try another hub. 

Care to share those 3 hubs/splitters that you have tried? So that I wouldn't buy the same.

tanjunior, 2021-11-27:
Update: tested with ONEPLUS 3T and discovered the same issue.

stevvie, 2021-11-27:
they are just unbranded ebay buys. I haven't heard of anybody that has a working usbC hub/cable that charges and does data. I have a micro usb hub that works on micro usb but a usbC one eludes me.

tanjunior, 2021-11-30:
@stevvie I guess USB-C protocols doesn't support this kind of usage. I read it some where while I did a google search. Can't find the link anymore. Something to do with host mode etc

closed 2023-03-23 by alufers