feelfreelinux/octo4a #95: Webcam support

Chrismaster2809, 2021-11-26:
Hey I love your application so far but is it possible to connect a Webcam to octo4a instead of the build in camera and if it is how can I do it.
Thank you for your support.👍

Gurbalur, 2022-07-18:
I have a MiniX Neo X7 running Android 5.1.1. I also have a Logitech C615 cam. Octo4a can find the camera. It calls it HD Webcam C615 but doesn't know which serial driver to use. I have selected all of them manually but no streaming so far.

lucky13820, 2022-08-27:
does anyone figure this out? I'm use a pixel 4 and C922, Octo4a can find the camera, but I can't get the streaming working.

stayingarmed, 2022-12-15:
Has anything come of this? I'm running this on a Galaxy Note 8 and have the phone camera working but I would like to add a second camera (Logitech C615) and it shows up when I plug it in, but I don't know how to go about getting it to show up in octoprint.

Gurbalur, 2022-12-15:
I gave up. Haven't tried in a few months.

sv0o, 2023-02-06:
I just installed octo4a 1.2 effortlessly on a Jide Remix Mini.  No ssh reinstalling necessary. I even did the Klipper support and OctoKlipper installs just to see if I could and those appear to be working even though I haven't flashed Klipper to my printer yet. 

I have a couple of USB webcams that I'd like to connect to the Remix, but it looks like that needs an octo4a extension/plugin maybe?

Korvorkian, 2023-03-22:
Also shaking the tree about getting more USB Webcam support.

Running on a Pixel 3XL with a Sabrent OTG USB splitter to a CR10SPRO and trying to run the 3DO Nozzle cam here: https://github.com/3DO-EU/nozzle-camera

would love to make this work without and happy to test code or whatever is needed to help!

***Note****
just saw the 1.2 update and will be trying that shortly.

out of curiosity, is the issue that O4A doesnt see the devices or that they need to be plugged in on boot or something else?

jovicassia, 2025-01-14:
no solution for USB camera (on TV BOX)?