feelfreelinux/octo4a #96: serialhdl.error: mcu 'mcu': Serial connection closed

GeeMing, 2021-11-27:
Hi, thanks for your efforts, this is a fantastic project!

but I having issue to connect the printer via USB serial port, running `tail /tmp/klippy.log` in ssh it shown:
```
localhost:~# tail /tmp/klippy.log 
    params = self.send_with_response(msg, 'identify_response')
  File "/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/klipper/klippy/serialhdl.py", line 315, in get_response
    self.serial.raw_send_wait_ack(cmds[-1], minclock, reqclock,
  File "/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
serialhdl.error: mcu 'mcu': Serial connection closed
```

then I kill the klipper service using "kill.sh" script, and manually run command:
```sh
python3 /klipper/klippy/klippy.py /root/printer.cfg -a /tmp/klippy_uds # without log path specified
```

then the klipper will show error in terminal:
```
INFO:root:mcu 'mcu': Starting serial connect
WARNING:root:mcu 'mcu': Unable to open serial port: [Errno 13] Could not exclusively lock port /dev/ttyOcto4a: [Errno 13] Permission denied
```

after that I try install minicom (apk add minicom), connect the /dev/ttyOcto4a with 
```sh
minicom -D /dev/ttyOcto4a -H -b 25000
```

I can see the minicom terminal printout the hex data which reported by Klipper firmware in mega2560 (I'd already flash the firmware via PC and test in PC environemt, which original octoprint can works).

Did I do something wrong? How can I solve this issue?

FYI:
- device: `Lenovo Zuk z2`
- os: aospextended v8.6 (`Android 11`)
- octo4a: `v1.0.4`

Mugesh-Rajaram, 2022-03-01:
Hi @GeeMing,

Can you tell me how to install/configure klipper with this **octo4a** mobile app.

GeeMing, 2022-03-05:
Hi Mugesh,

what I did to install/configure Klipper:
1. copy `scripts/setup-klipper.sh` into the smart phone via ssh
2. execute it inside the shell `bash setup-klipper.sh`
3. clone the `printer.cfg` into `/root/`, a worked Klipper configuration from my Klipper host control PC
4. change `printer.cfg` **[mcu]** section as `serial: /dev/ttyOcto4a`
5. restart the APP

sheminasalam, 2022-07-04:
> tail /tmp/klippy.log

Were you able to solve this? I am also facing the same issue.

thingsapart, 2022-08-11:
I had somewhat similar problems but prefixing klipper invocations with `LD_PRELOAD=/home/octoprint/ioctlHook.so python3 klipper.py ...`, e.g. update`start.sh` to:

```
LD_PRELOAD=/home/octoprint/ioctlHook.so  python3 /klipper/klippy/klippy.py /root/printer.cfg -l /tmp/klippy.log -a /tmp/klippy_uds
```