feelfreelinux/octo4a #98: Android TV box - ethernet connection

Woytacze, 2021-11-27:
I have old android TV box Schishison V88 with Android 5 and 5" touchscreen hdmi lcd  for Raspberry Pi.
I know that is not intended device but why not? There is lot of these old devices and even new ones are not expensive. So I tried it just for fun.
When i was testing it I was connected to my network by ethernet. App (1.0.4) instaled no problem even cheap USB webcam worked in preview. But ip adress of octoprint server was 0.0.0.0:5000. When I switched on wifi, server detected this connection and get wifi IP adress.
So more control over networks would be appreciated.
I gues that some people would like to use some USB Ethernet adapter a could have this issue.

IFIBSS, 2022-02-25:
same Problem got an Galaxy Tablet with an USB-C HUB and cant access the WebUI, when connected via the Ethernet Port of the Hub. WiFi is working well.

Ketzukhan, 2022-04-24:
Any solution? I'm having the same exact issue on an old s8 when I connect it to an ethernet cable

alufers, 2023-03-26:
Fixed in https://github.com/feelfreelinux/octo4a/commit/7d3cbd67ff94d27062a848e048876fd371c43864

closed 2023-03-26 by alufers