feelfreelinux/octo4a #104: Android 7 TV box

Meguinness, 2021-12-03:
Installed octo with no problems, running smoothly and quick , the problem comes when starting the camera server. 
Sometimes the app will force close even before connecting a camera , and sometimes will work ok but once I reboot the TV box the octo app will not start unless I go to the app and disable camera permission.
Any possible solution? Thank you

feelfreelinux, 2021-12-04:
Hi, is the camera a usb connected one?

Meguinness, 2021-12-04:
Hi yes it is us... ..but the app still force close even without the camera been connected . I tried different ways, .... booting up with camera inserted in the usb port ...or without..., About 1 minute later it fill force close.
I believe the problem is some how related to permissions.....in attached photo when I tick DONT ASK AGAIN the allow button becomes unselectable. Thank you

Meguinness, 2021-12-04:
[image]

Meguinness, 2021-12-04:
Hi yes it is us... ..but the app still force close even without the camera been connected . I tried different ways, .... booting up with camera inserted in the usb port ...or without..., About 1 minute later it fill force close.
I believe the problem is some how related to permissions.....in attached photo when I tick DONT ASK AGAIN the allow button becomes unselectable. Thank you
[image]

emailcraigs, 2022-05-23:
Did you ever get to the bottom of this? I'm having the same issue. Using USB cam. When enabling the Camera server it crashes and then I have to clear data and cache and reinstall before it will work again.

pestregen, 2022-12-01:
> Did you ever get to the bottom of this? I'm having the same issue. Using USB cam. When enabling the Camera server it crashes and then I have to clear data and cache and reinstall before it will work again.

I have the same issue with a Android 11 TV-Box.

Meguinness, 2022-12-01:
I was tired of the reliability issues so  I purchased a raspberry pi and using a hold Android phone as a IP webcam.

alufers, 2023-03-21:
It should be fixed in >=1.2.1. There was an issue when there were 0 cameras in the device.

closed 2023-03-21 by alufers