feelfreelinux/octo4a #105: Sony Xperia Z1 (5.0.2): Won't even install

MelodeathGuy, 2021-12-05:
Xperia Z1 is the only device that I have which can be charged while plugged into my printer. Sadly octo4a instantly crashes when I try to install it. (I have a LG G6 lying around and it works just fine, although it can't be charged so it is useless.)

[Here is the video shows how it crashes](https://youtube.com/shorts/sv85ZsNzdYc?feature=share) 

I'm on the latest build btw.

alufers, 2021-12-05:
I have screenshotted the logs from the video, for further invstigation

[image]

kobzzon, 2021-12-06:
I have a similar problem
[image]
![Screenshot_20211206-082044](https://user-images.githubusercontent.com/95584234/144786766-e8f3ef01-014c-48d6-9a7c-f9dd59fd39a2.png
[image]
)
[image]

motocykal, 2022-01-12:
I have a similar failure when trying to install dependencies on my Sony Xperia Z3Compact as well.

When I tap allow to enable the app access to storage space, it'll prompt that a screen overlay needs to be disabled. However, this prompt constantly appears even after I have disabled all apps with a screen overlay. It's like it's incorrectly recognising a screen overlay?

The dependencies install also fails even if I manually allow storage permissions before starting the process.

miroan81, 2022-02-20:
> Xperia Z1 is the only device that I have which can be charged while plugged into my printer. Sadly octo4a instantly crashes when I try to install it. (I have a LG G6 lying around and it works just [...]

Ive had the same problem with my Z1, you need to flash your phone to different fw and it will work. I flashed android 10 from xxda and this thing runs better than expected.

MelodeathGuy, 2022-02-20:
Been trying to flash a custom for a while now, but the compatibility between Windows 10 and fastboot drivers are trash. How did you unlock your bootloader?

miroan81, 2022-02-20:
> Been trying to flash a custom for a while now, but the compatibility between Windows 10 and fastboot drivers are trash. How did you unlock your bootloader?

I have an old laptop with win7 laying around just for these things
First check if your bootloader is unlockable here before doing anything
http://unlockbootloader.sonymobile.com/
I then used program named Flashtool to unlock it with the code Sony provided.
If you need further advice, let me know

iat31, 2022-03-08:
I had the same problem with installation on stock OS on Z1 Compact.
I installed latest version of Lineage OS from Feb 2022 and works fine now.

miroan81, 2022-03-09:
> I had the same problem with installation on stock OS on Z1 Compact. I installed latest version of Lineage OS from Feb 2022 and works fine now.

Im also running Lineage with Octoprint and Airdroid so I can remotely change things since my touch isn't working + my phone is now in case with active cooling (fan), it keeps temp low since the phone is overheating from day one I bought it. Octophone hasn't failed once so I call this operation a complete success