feelfreelinux/octo4a #112: Camera Focus Issue

outerpip, 2021-12-12:
Working on tweaking camera issues, and I haven't found a successful way to manually focus the camera, and while on autofocus, the smartphone camera is trying to focus every few movements of the extruder, so it gets a bit annoying. Have you worked with this any?

Thanks,

tsbrewers, 2022-04-10:
I am having the same issue and trying to figure something out, have you had any luck?

outerpip, 2022-04-10:
I've actually upgraded my device and no longer have camera issues. Assuming it was a hardware/software issue with the Galaxy S7. My LG G8's camera is perfect.

tsbrewers, 2022-04-10:
Thanks, I am using an old HTC 10 and if I leave focus on, you can see it focusing on the time lapse, but if I turn it off, it is always out of focus.   I'll keep tinkering.

outerpip, 2022-04-10:
That's exactly what mine was doing. Couldn't ever figure anything out. I ended up mounting my phone to the bed so it moved with the model, and that helped some, but the trade off was the shaking created by the movement.

DonEugraf, 2023-02-06:
If you are trying octolapse and move the head on the side before taking picture would it solve the problem?

joejtrsmoe, 2023-02-25:
Any chance there could be a slider below auto focus off to set focal length? Would have to be added to app?

DuyNK-3DP, 2023-06-29:
Hi @alufers ,
I'm someone who doesn't know anything about coding.
but I did my own research and found keywords on google about camera android focus distance and focus mode manually.
i don't know but i feel you can help check and add that command line. "CaptureRequest.LENS_FOCUS_DISTANCE, 0.0F"
` .setCaptureRequestOption(
                        CaptureRequest.CONTROL_AF_MODE,
                        CameraMetadata.CONTROL_AF_MODE_OFF
                    )
                    .setCaptureRequestOption(
                        CaptureRequest.LENS_FOCUS_DISTANCE, 0.0F
                    )`