feelfreelinux/octo4a #113: Gcc can't be used when plugins are installing

QuinnDamerell, 2021-12-12:
Hello! I'm the developer behind the OctoEverywhere plugin for OctoPrint. I love your system but have gotten a few support tickets about an issue with my plugin install. I can't tell at all how many users hit this issue, but I have gotten 4-5 tickets in the last two weeks or so one it.

The issue is that I use the brotli compression package with my plugin. When it tries to pip install one of the install processes tries to use gcc to compile native code. It seems on some installs of octo4a gcc isn't part of the system and it fails. Here's a OctoPrint log:

`2021-12-12 19:21:26,086 > Using legacy 'setup.py install' for brotli, since package 'wheel' is not installed.
2021-12-12 19:21:26,086 > Installing collected packages: brotli, OctoEverywhere
2021-12-12 19:21:26,087 > Running setup.py install for brotli: started
2021-12-12 19:21:30,785 ! ERROR: Command errored out with exit status 1:
2021-12-12 19:21:30,787 ! command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-pfj4ucyk/brotli_b1e03015eaab429fa3d5a6a0d37ff01d/setup.py'"'"'; __file__='"'"'/tmp/pip-install-pfj4ucyk/brotli_b1e03015eaab429fa3d5a6a0d37ff01d/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-nuiw99o0/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/brotli
2021-12-12 19:21:30,788 ! cwd: /tmp/pip-install-pfj4ucyk/brotli_b1e03015eaab429fa3d5a6a0d37ff01d/
2021-12-12 19:21:30,789 ! Complete output (15 lines):
2021-12-12 19:21:30,790 ! running install
2021-12-12 19:21:30,791 ! running build
2021-12-12 19:21:30,792 ! running build_py
2021-12-12 19:21:30,793 ! creating bin
2021-12-12 19:21:30,794 ! creating bin/lib.linux-aarch64-3.9
2021-12-12 19:21:30,795 ! copying python/brotli.py -> bin/lib.linux-aarch64-3.9
2021-12-12 19:21:30,796 ! running build_ext
2021-12-12 19:21:30,797 ! building '_brotli' extension
2021-12-12 19:21:30,798 ! creating bin/temp.linux-aarch64-3.9
2021-12-12 19:21:30,799 ! creating bin/temp.linux-aarch64-3.9/c
2021-12-12 19:21:30,800 ! creating bin/temp.linux-aarch64-3.9/c/common
2021-12-12 19:21:30,800 ! creating bin/temp.linux-aarch64-3.9/c/dec
2021-12-12 19:21:30,801 ! creating bin/temp.linux-aarch64-3.9/c/enc
2021-12-12 19:21:30,802 ! gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -g -fno-semantic-interposition -g -fno-semantic-interposition -g -fno-semantic-interposition -DTHREAD_STACK_SIZE=0x100000 -fPIC -Ic/include -I/usr/include/python3.9 -c c/common/constants.c -o bin/temp.linux-aarch64-3.9/c/common/constants.o
2021-12-12 19:21:30,803 ! error: command 'gcc' failed: No such file or directory
2021-12-12 19:21:30,804 ! ----------------------------------------
2021-12-12 19:21:30,805 ! ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-pfj4ucyk/brotli_b1e03015eaab429fa3d5a6a0d37ff01d/setup.py'"'"'; __file__='"'"'/tmp/pip-install-pfj4ucyk/brotli_b1e03015eaab429fa3d5a6a0d37ff01d/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-nuiw99o0/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/brotli Check the logs for full command output.
2021-12-12 19:21:31,315 > Running setup.py install for brotli: finished with status 'error'
`

I have seen OctoEverywhere work on some Android phones (I assume most?) so I'm not sure what the difference is. I also don't really know how Octo4A is setup, does it run in a container or how does it run python? Would the version of GCC it finds be something that's installed on the phone already? Or is it running in some kind of container where it's using a different version?

I'm going to try to look into how I can work around this in my plugin. Right now the only way I have found is to not have it as an install requirement, and then to try to pip install it via the plugin, which seems like it's not ideal in PY. 

I would love to know if you have any thoughts about how this could be fixed in Octo4A or in my plugin logic! Thanks!

QuinnDamerell, 2021-12-17:
Hey @feelfreelinux if you have a quick minute, can you share your thoughts on this? Do you think it's possible to fix this or is it most likely that the phones just don't have gcc installed and thus won't really ever work? I'm trying to come up with a workaround in my logic, but haven't thought of anything good yet. :(

alufers, 2021-12-18:
I think you need to enable "Plugin Extras" in the settings so that gcc is installed.

Snicket8247, 2021-12-24:
@alufers I do have Plugin Extras installed but it's not working so it might be something else.

Snicket8247, 2021-12-24:
@QuinnDamerell I'm on of the people who contacted you about it and after not working for a while I just tried installing OctoEverywhere again and it worked this time.

Really no idea what have changed.
Maybe a couple restarts of the server did the trick.

QuinnDamerell, 2021-12-25:
@Snicket8247 good to hear it’s working! It was two things:

1) @alufers is correct and you need to enable the plug-in extras
2) Since I didn't want users to be confused, I just removed the dependency that needed gcc from the plug-in! So going forward the plug-in should just work for everyone!

closed 2021-12-25 by QuinnDamerell