feelfreelinux/octo4a #119: App crashes when "INSTALL OCTOPRINT" pressed

ncrmnt, 2021-12-16:
I tred to install octo4a on my old tablet Acer A500 (Android 5.1.1), but without success. Everytime when I press "INSTALL OCTOPRINT" app crashes.

App version 1.0.4

App permissions
[Screenshot-2021-12-16-16-49-00-copy.png](https://postimg.cc/1V18hWbk)

logcat:
```
12-16 16:02:31.869   551   591 I ActivityManager: Displayed com.octo4a/.ui.InitialActivity: +1s118ms
12-16 16:02:31.869   551   591 I Timeline: Timeline: Activity_windows_visible id: ActivityRecord{12dc104d u0 com.octo4a/.ui.InitialActivity t29} time:3572491
12-16 16:02:31.886  8418  8418 I Timeline: Timeline: Activity_idle id: android.os.BinderProxy@3859f794 time:3572508
12-16 16:02:31.987   551  1147 D TaskPersister: removeObsoleteFile: deleting file=28_task.xml
12-16 16:03:02.900  1774  8444 W Uploader: No account for auth token provided
12-16 16:03:02.912  1774  8444 I PseudonymousIdIntentService: getToken invoked
12-16 16:03:03.357  8418  8418 I Timeline: Timeline: Activity_launch_request id:com.octo4a time:3603978
12-16 16:03:03.357   551  1312 I ActivityManager: START u0 {cmp=com.octo4a/.ui.InstallationActivity} from uid 10073 on display 0
12-16 16:03:03.365   551  1312 V WindowManager: addAppToken: AppWindowToken{1c76ebac token=Token{1905855f ActivityRecord{20ba79fe u0 com.octo4a/.ui.InstallationActivity t29}}} to stack=1 task=29 at 1
12-16 16:03:03.523  8418  8418 W linker  : libvsp-pty.so: unused DT entry: type 0x6ffffef5 arg 0x86c
12-16 16:03:03.523  8418  8418 W linker  : libvsp-pty.so: unused DT entry: type 0x6ffffffe arg 0xb00
12-16 16:03:03.523  8418  8418 W linker  : libvsp-pty.so: unused DT entry: type 0x6fffffff arg 0x2
12-16 16:03:03.524  8418  8418 V GetEnv: :  Subscribe to Listener  OK 
12-16 16:03:03.526  8418  8448 V TAG     : PtyThread getting ready
12-16 16:03:03.526  8418  8448 V TAG     : PtyThread called openpty
12-16 16:03:03.526  8418  8448 V TAG     : SYMLIKED
--------- beginning of crash
12-16 16:03:03.526  8418  8448 F libc    : Fatal signal 4 (SIGILL), code 1, fault addr 0xa4a663ca in tid 8448 (Thread-966)
12-16 16:03:03.526  7259  7259 I DEBUG   : property debug.db.uid not set; NOT waiting for gdb.
12-16 16:03:03.526  7259  7259 I DEBUG   : HINT: adb shell setprop debug.db.uid 100000
12-16 16:03:03.526  7259  7259 I DEBUG   : HINT: adb forward tcp:5039 tcp:5039
12-16 16:03:03.563  8418  8418 V GetEnv: :  Subscribe to Listener  OK 
12-16 16:03:03.628  7259  7259 I DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
12-16 16:03:03.628  7259  7259 I DEBUG   : Build fingerprint: 'acer/a500_ww_gen1/picasso:5.1.1/LMY48B/1333032611:user/release-keys'
12-16 16:03:03.628  7259  7259 I DEBUG   : Revision: '0'
12-16 16:03:03.628  7259  7259 I DEBUG   : ABI: 'arm'
12-16 16:03:03.628  7259  7259 I DEBUG   : pid: 8418, tid: 8448, name: Thread-966  >>> com.octo4a <<<
12-16 16:03:03.628  7259  7259 I DEBUG   : signal 4 (SIGILL), code 1 (ILL_ILLOPC), fault addr 0xa4a663ca
12-16 16:03:03.640  7259  7259 I DEBUG   :     r0 00002086  r1 00005420  r2 a43ffb54  r3 a43ff908
12-16 16:03:03.641  7259  7259 I DEBUG   :     r4 a43ffc5c  r5 a4a68410  r6 a4a683db  r7 a43ffd90
12-16 16:03:03.642  7259  7259 I DEBUG   :     r8 a440ee88  r9 00000000  sl a4a6a004  fp a43ffdb0
12-16 16:03:03.643  7259  7259 I DEBUG   :     ip a43ffd90  sp a43ff910  lr a4a663c9  pc a4a663ca  cpsr 000e0030
12-16 16:03:03.643  7259  7259 I DEBUG   : 
12-16 16:03:03.643  7259  7259 I DEBUG   : backtrace:
12-16 16:03:03.643  7259  7259 I DEBUG   :     #00 pc 000013ca  /data/app/com.octo4a-1/lib/arm/libvsp-pty.so
12-16 16:03:03.643  7259  7259 I DEBUG   :     #01 pc 000131c7  /system/lib/libc.so (__pthread_start(void*)+30)
12-16 16:03:03.643  7259  7259 I DEBUG   :     #02 pc 0001113f  /system/lib/libc.so (__start_thread+6)
12-16 16:03:04.238   551  8457 W ActivityManager:   Force finishing activity 1 com.octo4a/.ui.InstallationActivity
12-16 16:03:04.239  7259  7259 I DEBUG   : 
12-16 16:03:04.239  7259  7259 I DEBUG   : Tombstone written to: /data/tombstones/tombstone_08
12-16 16:03:04.254   551  1570 I WindowState: WIN DEATH: Window{1df4f067 u0 com.octo4a/com.octo4a.ui.InitialActivity}
12-16 16:03:04.260   551  8457 W ActivityManager: Exception thrown during pause
12-16 16:03:04.260   551  8457 W ActivityManager: android.os.DeadObjectException
12-16 16:03:04.260   551  8457 W ActivityManager: 	at android.os.BinderProxy.transactNative(Native Method)
12-16 16:03:04.260   551  8457 W ActivityManager: 	at android.os.BinderProxy.transact(Binder.java:496)
12-16 16:03:04.260   551  8457 W ActivityManager: 	at android.app.ApplicationThreadProxy.schedulePauseActivity(ApplicationThreadNative.java:704)
12-16 16:03:04.260   551  8457 W ActivityManager: 	at com.android.server.am.ActivityStack.startPausingLocked(ActivityStack.java:832)
12-16 16:03:04.260   551  8457 W ActivityManager: 	at com.android.server.am.ActivityStack.finishActivityLocked(ActivityStack.java:2827)
12-16 16:03:04.260   551  8457 W ActivityManager: 	at com.android.server.am.ActivityStack.finishTopRunningActivityLocked(ActivityStack.java:2684)
12-16 16:03:04.260   551  8457 W ActivityManager: 	at com.android.server.am.ActivityStackSupervisor.finishTopRunningActivityLocked(ActivityStackSupervisor.java:2525)
12-16 16:03:04.260   551  8457 W ActivityManager: 	at com.android.server.am.ActivityManagerService.handleAppCrashLocked(ActivityManagerService.java:11814)
12-16 16:03:04.260   551  8457 W ActivityManager: 	at com.android.server.am.ActivityManagerService.makeAppCrashingLocked(ActivityManagerService.java:11686)
12-16 16:03:04.260   551  8457 W ActivityManager: 	at com.android.server.am.ActivityManagerService.crashApplication(ActivityManagerService.java:12395)
12-16 16:03:04.260   551  8457 W ActivityManager: 	at com.android.server.am.ActivityManagerService.handleApplicationCrashInner(ActivityManagerService.java:11906)
12-16 16:03:04.260   551  8457 W ActivityManager: 	at com.android.server.am.NativeCrashListener$NativeCrashReporter.run(NativeCrashListener.java:86)
12-16 16:03:04.262   120   120 E lowmemorykiller: Error writing /proc/8418/oom_score_adj; errno=22
12-16 16:03:04.269   551  1501 I ActivityManager: Process com.octo4a (pid 8418) has died
12-16 16:03:04.270   551  1501 W ActivityManager: Scheduling restart of crashed service com.octo4a/.service.OctoPrintService in 1000ms
12-16 16:03:04.319   551   584 V ActivityManager: killProcessGroupAsync took 50 ms for PID 8418 on thread 17
12-16 16:03:04.319   132   132 I Zygote  : Process 8418 exited due to signal (4)
12-16 16:03:04.358   551   734 I OpenGLRenderer: Initialized EGL, version 1.4
[... 13 lines cut ...]
12-16 16:03:05.761  8460  8460 V GetEnv: :  Subscribe to Listener  OK 
12-16 16:03:05.767  8460  8478 V TAG     : PtyThread getting ready
12-16 16:03:05.767  8460  8478 V TAG     : PtyThread called openpty
12-16 16:03:05.767  8460  8478 V TAG     : SYMLIKED
12-16 16:03:05.767  8460  8478 F libc    : Fatal signal 4 (SIGILL), code 1, fault addr 0xa51c33ca in tid 8478 (Thread-968)
12-16 16:03:05.768  7259  7259 I DEBUG   : property debug.db.uid not set; NOT waiting for gdb.
12-16 16:03:05.768  7259  7259 I DEBUG   : HINT: adb shell setprop debug.db.uid 100000
12-16 16:03:05.768  7259  7259 I DEBUG   : HINT: adb forward tcp:5039 tcp:5039
12-16 16:03:05.845  8460  8460 V GetEnv: :  Subscribe to Listener  OK 
12-16 16:03:05.870  7259  7259 I DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
12-16 16:03:05.870  7259  7259 I DEBUG   : Build fingerprint: 'acer/a500_ww_gen1/picasso:5.1.1/LMY48B/1333032611:user/release-keys'
12-16 16:03:05.870  7259  7259 I DEBUG   : Revision: '0'
12-16 16:03:05.870  7259  7259 I DEBUG   : ABI: 'arm'
12-16 16:03:05.871  7259  7259 I DEBUG   : pid: 8460, tid: 8478, name: Thread-968  >>> com.octo4a <<<
12-16 16:03:05.871  7259  7259 I DEBUG   : signal 4 (SIGILL), code 1 (ILL_ILLOPC), fault addr 0xa51c33ca
12-16 16:03:05.890  7259  7259 I DEBUG   :     r0 00002086  r1 00005420  r2 a51bdb54  r3 a51bd908
12-16 16:03:05.890  7259  7259 I DEBUG   :     r4 a51bdc5c  r5 a51c5410  r6 a51c53db  r7 a51bdd90
12-16 16:03:05.890  7259  7259 I DEBUG   :     r8 afc4fa88  r9 00000000  sl a51c7004  fp a51bddb0
12-16 16:03:05.890  7259  7259 I DEBUG   :     ip a51bdd90  sp a51bd910  lr a51c33c9  pc a51c33ca  cpsr 000e0030
12-16 16:03:05.891  7259  7259 I DEBUG   : 
12-16 16:03:05.891  7259  7259 I DEBUG   : backtrace:
12-16 16:03:05.891  7259  7259 I DEBUG   :     #00 pc 000013ca  /data/app/com.octo4a-1/lib/arm/libvsp-pty.so
12-16 16:03:05.891  7259  7259 I DEBUG   :     #01 pc 000131c7  /system/lib/libc.so (__pthread_start(void*)+30)
12-16 16:03:05.891  7259  7259 I DEBUG   :     #02 pc 0001113f  /system/lib/libc.so (__start_thread+6)
```

Marosk22, 2022-04-17:
Hi, I had the same problem. Go to App settings and move installation to Internal storage, then run Octo again and press Install Octoprint, worked for me. Hope it helps.

sresam89, 2022-05-21:
Should we put a disclosure to the install or move the installation to internal storage by default?
Closing the issue otherwise would be desired @feelfreelinux