feelfreelinux/octo4a #120: Print Freezes Mid Print - Connection to Printer Lost

FN3090, 2021-12-16:
Searched high and low to try to find a solution to this issue. Everything is working great with one exception - During the middle of a print, the print randomly freezes and in turn, creates a blob/zit on the print. I have tried changing all settings, USB printer cable, MAX/Min print resolution, wired and wireless phone connection, connection timeout settings...etc. Almost everything is working perfectly, except this one last issue.

In the terminal, it throws a lost connection with the printer error in the log when the print head freezes. It pauses for less than a second (typically), then resumes. It seems to happen randomly, but on a 30 minute print, seems to happen 5-6 times during the print.

I am using an Galaxy S10+. My printer is an FLSUN Super Racer with a Robbin Nano v3 board.

I love this project and really want to use it. Any suggestions?

Meguinness, 2021-12-20:
I'm having same problem on my android TV box....

Snicket8247, 2022-01-26:
Had the same issue occasionally most recently when switching from Cura to Prusa slicer.

I did a lot of playing around and testing and what fixed it for me was:
1. Install the "Arc welder" plugin.
2. Avoid using "curvy" infill like gyroids.
3. Not using z-hop in prusa slicer.