feelfreelinux/octo4a #123: Juttery Stuttery Prints.

B1gJ1mmy, 2021-12-20:
Love this software and I have found that an LG G3 has many of the features needed to get this to work well.  However I am having issues when printing especially outlines that go around bends etc.  The print stutters. This seems to line up with more complicated movements. 

I have tried several settings one the phone to try an boost performance e.g. powersaving keeping the screen on etc. However it still seems to be slow and stuttery. 

Anything that can help impove performace apart from getting a newer faster device etc.  I feel like these devices are perfect they hare wireless charging, a replacable abttery and are resonably modern.

Meguinness, 2021-12-20:
I'm having same problem it seams that my prints turned up with blobs on it when using the Octoprint app become it stops randomly for a split second enough to create the extrusion blobs... transmission problems??....but when printing from SD using the same gcode every print is without any problems. I think we really need v1.5 with the hope of squashed bugs👍

Snicket8247, 2021-12-24:
Having the same issue.

joseph-montanez, 2022-01-01:
Do you have the camera enabled? If so try disabling it.

Snicket8247, 2022-01-01:
I figured it was all about curves for me.
I'm using the triangle instead of the gyroid infill now and also use the "arc welder" plugin for parts that have a lot of curves.

Didn't have any issues since then.

B1gJ1mmy, 2022-01-01:
> Do you have the camera enabled? If so try disabling it. Also if you have any extra OctoPrint plugins, disable those as well.

Yes I have the camera enabled I want to monitor and do time-lapse. I have reduced to the minimum camera fps but still having the issue.

B1gJ1mmy, 2022-01-01:
> I figured it was all about curves for me. I'm using the triangle instead of the gyroid infill now and also use the "arc welder" plugin for parts that have a lot of curves.
> 
> Didn't have any issues since then.

Glad you got yours working.  The part I'm printing is cylindrical so avoiding curves is not an option.

Snicket8247, 2022-01-01:
> Glad you got yours working.  The part I'm printing is cylindrical so avoiding curves is not an option.

Then I highly recommend you still give the "Arc welder" plugin a shot.
It converts your G1 movements into G2/3 for real curves instead of multiple straight lines.

This not only fixes the blobs on the surface when the printer runs out of commands but also increases the quality of the surface in my experience.

B1gJ1mmy, 2022-01-03:
> Then I highly recommend you still give the "Arc welder" plugin a shot. It converts your G1 movements into G2/3 for real curves instead of multiple straight lines.
>
> This not only fixes the blobs on [...]

Thanks.  I recently tried to install arc welder.  I just get errors. I dont really have time to troubleshoot too much. 

ERROR: Command errored out with exit status 1:
command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-me72h3kj/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-me72h3kj/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-8o3c2rk3/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/Arc-Welder
cwd: /tmp/pip-req-build-me72h3kj/
Complete output (96 lines):
Unknown Version, falling back to 1.0.0+u.bb71e8f.
Found packages: {'octoprint_arc_welder', 'octoprint_arc_welder_setuptools'}
running install
running build
running build_py
creating build
creating build/lib.linux-armv7l-3.9
creating build/lib.linux-armv7l-3.9/octoprint_arc_welder
copying octoprint_arc_welder/__init__.py -> build/lib.linux-armv7l-3.9/octoprint_arc_welder
copying octoprint_arc_welder/_version.py -> build/lib.linux-armv7l-3.9/octoprint_arc_welder
copying octoprint_arc_welder/log.py -> build/lib.linux-armv7l-3.9/octoprint_arc_welder
copying octoprint_arc_welder/preprocessor.py -> build/lib.linux-armv7l-3.9/octoprint_arc_welder
copying octoprint_arc_welder/utilities.py -> build/lib.linux-armv7l-3.9/octoprint_arc_welder
creating build/lib.linux-armv7l-3.9/octoprint_arc_welder_setuptools
copying octoprint_arc_welder_setuptools/__init__.py -> build/lib.linux-armv7l-3.9/octoprint_arc_welder_setuptools
running egg_info
creating Arc_Welder.egg-info
writing Arc_Welder.egg-info/PKG-INFO
writing dependency_links to Arc_Welder.egg-info/dependency_links.txt
writing entry points to Arc_Welder.egg-info/entry_points.txt
writing requirements to Arc_Welder.egg-info/requires.txt
writing top-level names to Arc_Welder.egg-info/top_level.txt
writing manifest file 'Arc_Welder.egg-info/SOURCES.txt'
reading manifest file 'Arc_Welder.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching 'README.md'
writing manifest file 'Arc_Welder.egg-info/SOURCES.txt'
creating build/lib.linux-armv7l-3.9/octoprint_arc_welder/data
creating build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib
creating build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c
creating build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/arc_welder
copying octoprint_arc_welder/data/lib/c/arc_welder/arc_welder.cpp -> build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/arc_welder
copying octoprint_arc_welder/data/lib/c/arc_welder/segmented_arc.cpp -> build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/arc_welder
copying octoprint_arc_welder/data/lib/c/arc_welder/segmented_shape.cpp -> build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/arc_welder
creating build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/gcode_processor_lib
copying octoprint_arc_welder/data/lib/c/gcode_processor_lib/array_list.cpp -> build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/gcode_processor_lib
copying octoprint_arc_welder/data/lib/c/gcode_processor_lib/circular_buffer.cpp -> build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/gcode_processor_lib
copying octoprint_arc_welder/data/lib/c/gcode_processor_lib/extruder.cpp -> build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/gcode_processor_lib
copying octoprint_arc_welder/data/lib/c/gcode_processor_lib/gcode_comment_processor.cpp -> build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/gcode_processor_lib
copying octoprint_arc_welder/data/lib/c/gcode_processor_lib/gcode_parser.cpp -> build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/gcode_processor_lib
copying octoprint_arc_welder/data/lib/c/gcode_processor_lib/gcode_position.cpp -> build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/gcode_processor_lib
copying octoprint_arc_welder/data/lib/c/gcode_processor_lib/logger.cpp -> build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/gcode_processor_lib
copying octoprint_arc_welder/data/lib/c/gcode_processor_lib/parsed_command.cpp -> build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/gcode_processor_lib
copying octoprint_arc_welder/data/lib/c/gcode_processor_lib/parsed_command_parameter.cpp -> build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/gcode_processor_lib
copying octoprint_arc_welder/data/lib/c/gcode_processor_lib/position.cpp -> build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/gcode_processor_lib
copying octoprint_arc_welder/data/lib/c/gcode_processor_lib/utilities.cpp -> build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/gcode_processor_lib
creating build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/py_arc_welder
copying octoprint_arc_welder/data/lib/c/py_arc_welder/py_arc_welder.cpp -> build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/py_arc_welder
copying octoprint_arc_welder/data/lib/c/py_arc_welder/py_arc_welder_extension.cpp -> build/lib.linux-armv7l-3.9/octoprint_arc_welder/data/lib/c/py_arc_welder
[... cut ...]
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-me72h3kj/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-me72h3kj/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-8o3c2rk3/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/Arc-Welder Check the logs for full command output.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

joseph-montanez, 2022-01-03:
Make sure to install "Extras" in the Android app (its in the settings), this will install `GCC`. Restart Octoprint just in case.

B1gJ1mmy, 2022-01-09:
> Make sure to install "Extras" in the Android app (its in the settings), this will install `GCC`. Restart Octoprint just in case.

That's seems to be working. Pretty sure I had that enabled before but must have forgot when I reinstalled oct4a recently.

B1gJ1mmy, 2022-01-09:
> Make sure to install "Extras" in the Android app (its in the settings), this will install `GCC`. Restart Octoprint just in case.

That looks like it has fixed it.  

> Then I highly recommend you still give the "Arc welder" plugin a shot. It converts your G1 movements into G2/3 for real curves instead of multiple straight lines.
>
> This not only fixes the blobs on [...]

That looks like it's done the Job.  Just got to remeber to use the plugin now.  

Thanks to all that helped me out.

Snicket8247, 2022-01-26:
Just a follow up:
I eliminated the issue for me with the "arc welder" plugin and by avoiding "curvy" infill like gyroid as I mentioned before.

Now when I switched to Prusa slicer it started happening again and I got rid of ti by not using z-hop.

alufers, 2022-10-07:
Unfortunately there is really nothing octo4a can do to fix stuttering :/

closed 2022-10-07 by alufers