feelfreelinux/octo4a #128: Not connecting to BTT SKR Mini E3 via OTG cable

MsaterSgnt, 2021-12-28:
I'm trying to setup octo4a, but it keeps telling me that the printer is not connected. After looking at the FAQ, I saw a section for that there, but I'm not sure which option to use or how to implement it.

MsaterSgnt, 2021-12-29:
It now tells me that it is connected via the cable, but the octoprint webpage will not connect to the printer

joseph-montanez, 2022-01-01:
I got it working using the CDC option from the list of manual USB connections options. While it reports some /dev/usb/... it should still work. I'd recommend restarting Octoprint after you set the manual USB connection just in case. 

Do not if you have a complex OTG host with things like HDMI, Ethernet, etc you'll see a bunch of USB connection options, so you'll need to weed out which one is really for the USB port versus HDMI.

SjamonDaal, 2022-01-22:
Works on my printer but it requires me to first turn on the printer and then connect it via OTG.

eggandwaffles, 2022-04-02:
Changing the driver to CDC worked great for me as well, with no reboots or anything required. Thank you!

manningcustom, 2022-09-02:
> 
can you explain what the cdc connection is?

closed 2022-10-06 by alufers