feelfreelinux/octo4a #130: Serial connection issue with FF Creator Pro

RayLSmith, 2021-12-30:
Phone states it's connected and on dev/bus/002/002

GPX brings up Replicator options as well as the standard  Erikson one.

Serial port defined in OP is /dev/tty/Octo4a     Buad Rate: 11520    Printer profile: opserver

**Terminal log shows:**
Changing monitoring state from "Offline" to "Opening serial connection"
Changing monitoring state from "Opening serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
Unexpected error while connecting to serial port /dev/ttyOcto4a, baudrate 115200 from hook GPX: Timeout: 'Timeout' @ comm.py:_open_serial:3775

Both the Erikson and the Replicator have multiple choices and am unsure which I should use.

Any help would be most apprecialted.

Thanks,

Ray

Marmadukethe1st, 2022-01-14:
Same Problems for me

brolly916, 2022-02-24:
any one solved this issue as  I have the same issue too , I am using an old android box  (with fill size usb ports) and not able to connect / access my CR10s5 :/ i managed to choose the /dev/ttyOcto4a. however octoprint is still not able to connect / access my printer

can you kindly advise which driver I need to choose for my CR10 S5 please

solo2424, 2022-07-31:
I am trying to figure out what driver to use for my Makerbot Replicator 2X. Can anyone assist?

mangamike, 2022-11-09:
I am also having this issue. Using a Pixel 3 running Android 12 everything is running smoothly camera works, plugins all installed including GPX, but serial connection is not working on Flashforge Creator Pro. Octo4a see my printer as The Replicator, but I am unsure as to which serial driver to use i tried CDC/ACM which should be correct, to no avail. keeps erroring out and saying No more candidates to test, and no working port/baudrate combination detected.  https://imgur.com/a/3K8R52G

Deevo22, 2022-12-31:
Has anyone managed to get a working connection using octo4a and gpx?  I have tried all combinations of driver type and baud rate with no luck... Its as if gpx is not hooking into the serial connection properly.

agilewebsolutions, 2024-09-10:
Same here, has anyone managed to solve this?