feelfreelinux/octo4a #141: Selecting a different back camera doesn't work

phamtom009, 2022-01-21:
Hello,

I'm running an LG G6 with two back cameras. Octo4a detects the two cameras, but selecting either one yield the same camera lens.  I've been trying to switch to the wide angle camera lens.

arnauavj, 2022-01-29:
I have the same problem with LG G6.

robinkin9, 2022-02-01:
Same problem with Xiaomi mi 9t, has three back cameras but only detects one front and one back camera.

Diiescher, 2023-06-23:
Hi, 
I just installed Octo4a 1.2.4 on a Samsung A13, having 
Back cameras: Macro 2MP, UW 5MP and main 50MP (F1.8) 
Front Camera: 8MP
in Octo4a I can select Back: 12MP and Front: 5 MP and 8 MP.
(both choices for front give the same front camera picture)

aaand it would be nice if we could select the cam and light on octoprint UI... just dreaming along ;)

btw: awesome project!!