feelfreelinux/octo4a #144: Can't install plugins

Rettatata, 2022-01-23:
I can install bed visualizer, but nothing else. Updating my pip from v20 to v21 also breaks my server and doesn't allow me to even try to install any more plugins. I'm guessing it's an issue with this server not being up to date and the plugins needing an updated pip? 

Here's exactly what the final message says
Installing plugin "Octolapse" from https://github.com/FormerLurker/Octolapse/archive/master.zip...
/usr/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpn_v53mu_/Octolapse-master.zip --no-cache-dir
Processing /tmp/tmpn_v53mu_/Octolapse-master.zip
Requirement already satisfied: pillow>=6.2.0<7.0.0 in /usr/lib/python3.9/site-packages (from Octolapse==0.4.1) (8.2.0)
Requirement already satisfied: sarge in /usr/lib/python3.9/site-packages (from Octolapse==0.4.1) (0.1.6)
Requirement already satisfied: six in /usr/lib/python3.9/site-packages (from Octolapse==0.4.1) (1.15.0)
Requirement already satisfied: OctoPrint>1.3.8 in /usr/lib/python3.9/site-packages (from Octolapse==0.4.1) (1.7.3)
Requirement already satisfied: psutil in /usr/lib/python3.9/site-packages (from Octolapse==0.4.1) (5.8.0)
Collecting file_read_backwards
Downloading file_read_backwards-2.0.0-py2.py3-none-any.whl (9.0 kB)
Requirement already satisfied: setuptools>=6.0 in /usr/lib/python3.9/site-packages (from Octolapse==0.4.1) (52.0.0)
Collecting awesome-slugify<1.7,>=1.6.5
Downloading awesome-slugify-1.6.5.tar.gz (8.4 kB)
Requirement already satisfied: regex in /usr/lib/python3.9/site-packages (from awesome-slugify<1.7,>=1.6.5->Octolapse==0.4.1) (2021.4.4)
Collecting Unidecode<0.05,>=0.04.14
Downloading Unidecode-0.04.21-py2.py3-none-any.whl (228 kB)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (2021.2.23)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (2021.10.11)
Requirement already satisfied: OctoPrint-PiSupport>=2021.10.28 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (2021.10.28)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.1.1)
Requirement already satisfied: markdown<3.2,>=3.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (3.1.1)
Requirement already satisfied: wrapt<1.13,>=1.12.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.12.1)
Requirement already satisfied: flask<2,>=1.1.4 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.1.4)
Requirement already satisfied: Jinja2<3,>=2.11.3 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (2.11.3)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (0.5.0)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.0.0)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (2.0)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.0.1)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.1.0)
Requirement already satisfied: cachelib<0.2,>=0.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (0.1.1)
Requirement already satisfied: PyYAML<6,>=5.4.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (5.4.1)
Requirement already satisfied: pyserial<4,>=3.4 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (3.5)
Requirement already satisfied: netaddr<0.9,>=0.8 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (0.8.0)
Requirement already satisfied: watchdog==0.10.4 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (0.10.4)
Requirement already satisfied: netifaces<1,>=0.11 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (0.11.0)
Requirement already satisfied: pylru<2,>=1.2 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.2.0)
Requirement already satisfied: pkginfo<2,>=1.7.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.8.2)
Requirement already satisfied: requests<3,>=2.26.0 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (2.27.1)
Requirement already satisfied: semantic_version<3,>=2.8.5 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (2.8.5)
Requirement already satisfied: Click<8,>=7.1.2 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (7.1.2)
Requirement already satisfied: future<1,>=0.18.2 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (0.18.2)
Requirement already satisfied: websocket-client<1,>=0.59 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (0.59.0)
Requirement already satisfied: emoji<2,>=1.4.2 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.6.3)
Requirement already satisfied: sentry-sdk<2,>=1.3.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.5.3)
Requirement already satisfied: filetype<2,>=1.0.7 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.0.9)
Requirement already satisfied: zipstream-new<1.2,>=1.1.8 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.1.8)
Requirement already satisfied: blinker<2,>=1.4 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.4)
Requirement already satisfied: feedparser<7,>=6.0.8 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (6.0.8)
Requirement already satisfied: tornado<7,>=6 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (6.1)
Requirement already satisfied: zeroconf<0.34,>=0.33 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (0.33.4)
Requirement already satisfied: immutabledict<3,>=2.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (2.2.1)
Requirement already satisfied: pathvalidate<3,>=2.4.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (2.5.0)
Requirement already satisfied: colorlog<6,>=5.0.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (5.0.1)
Requirement already satisfied: pathtools>=0.1.1 in /usr/lib/python3.9/site-packages (from watchdog==0.10.4->OctoPrint>1.3.8->Octolapse==0.4.1) (0.1.2)
Requirement already satisfied: sgmllib3k in /usr/lib/python3.9/site-packages (from feedparser<7,>=6.0.8->OctoPrint>1.3.8->Octolapse==0.4.1) (1.0.0)
Requirement already satisfied: webassets>=2.0 in /usr/lib/python3.9/site-packages (from Flask-Assets<3,>=2.0->OctoPrint>1.3.8->Octolapse==0.4.1) (2.0)
Requirement already satisfied: Babel>=2.3 in /usr/lib/python3.9/site-packages (from Flask-Babel<2,>=1.0->OctoPrint>1.3.8->Octolapse==0.4.1) (2.9.1)
Requirement already satisfied: pytz in /usr/lib/python3.9/site-packages (from Flask-Babel<2,>=1.0->OctoPrint>1.3.8->Octolapse==0.4.1) (2021.3)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /usr/lib/python3.9/site-packages (from requests<3,>=2.26.0->OctoPrint>1.3.8->Octolapse==0.4.1) (1.26.5)
Requirement already satisfied: certifi>=2017.4.17 in /usr/lib/python3.9/site-packages (from requests<3,>=2.26.0->OctoPrint>1.3.8->Octolapse==0.4.1) (2020.12.5)
Requirement already satisfied: idna<4,>=2.5 in /usr/lib/python3.9/site-packages (from requests<3,>=2.26.0->OctoPrint>1.3.8->Octolapse==0.4.1) (3.2)
Requirement already satisfied: charset-normalizer~=2.0.0 in /usr/lib/python3.9/site-packages (from requests<3,>=2.26.0->OctoPrint>1.3.8->Octolapse==0.4.1) (2.0.10)
Requirement already satisfied: ifaddr>=0.1.7 in /usr/lib/python3.9/site-packages (from zeroconf<0.34,>=0.33->OctoPrint>1.3.8->Octolapse==0.4.1) (0.1.7)
Using legacy 'setup.py install' for Octolapse, since package 'wheel' is not installed.
Using legacy 'setup.py install' for awesome-slugify, since package 'wheel' is not installed.
[... cut ...] open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-0gt4lp8j/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/Octolapse Check the logs for full command output.
Running setup.py install for Octolapse: finished with status 'error'
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

WRH0530, 2022-04-19:
I have the same problem

alufers, 2022-10-07:
Please use the "Install plugin extras" option from the settings before installing the plugins:

[image]

closed 2022-10-07 by alufers