feelfreelinux/octo4a #145: cannot connect phone to printer

benj04800, 2022-01-25:
I am using a ZTE 7 mini to connect to an Ender 5 pro the phone has USB OTG. When I attach the USB cable using the splitter it says that the printer is not connected. I think the issue is that the printer is not being automatically detected. I do not know how to manually select the correct driver which should be CH341 in the app.

benj04800, 2022-01-25:
update

I discovered that the issue is that when I have both the charging cable and the printer connected to the phone, the phone will not recognize the printer. When the charging cable is unplugged the phone will recognize the printer. How do I both charge and connect to the printer at the same time?

dangel1967, 2022-01-26:
I can't connect either. How exactly are we supposed to connect? The install went great till that happened.

Raul-7-7, 2022-01-29:
@benj04800 Do you have Wireless Charging on your device?

dangel1967, 2022-01-29:
No

Jacob12138xieyuan, 2022-02-10:
i encounter the same issue with huawei and ender3 v2

manningcustom, 2022-02-10:
I have it running on my tablet.  The server is running and it says the printer is connected.   But when I go to octoprint in my browser it will not connect and just  times out.   I have switched from auto to manual and tried every listed baudrate.   no luck

What am I missing?

miroan81, 2022-02-20:
> 

i soldered 2 wires directly to my battery which are then connected to external charging module tp4056. By adding that you technically have 2 usb connectors, one for charging and other one for OTG.

lola66, 2022-03-08:
> update
>
> I discovered that the issue is that when I have both the charging cable and the printer connected to the phone, the phone will not recognize the printer. When the charging cable is unplugged [...]

I'm having the same issue with several USB C adapters and 2 recent android devices... as soon as I plug in the power adapter the serial connection is ended... I disconnect the power the serial connection is back.. very disappointing. I haven't found a solution/adapter that works. 

I cannot use the ocot4a app/phone instead of an rpi.

Raul-7-7, 2022-03-08:
I been using S7 Edge Samsung with wireless charging and OTG cable for months now without issue. If your phone has wireless charging, give that a shot.

lola66, 2022-03-09:
> I been using S7 Edge Samsung with wireless charging and OTG cable for months now without issue. If your phone has wireless charging, give that a shot.

Sadly they don't, (Umidigi Power and F1) If wireless charging is the "easy" fix, fine. But that information is missing from the main Github project page. An explanation or warning about this major hardware limitation should be included IMO. If the phone can't be charged while printing then it's not a viable solution for running octoprint for hours/days...   I do like the idea or recycling my old phones, so I will look into removing the batteries and powering them directly.

miroan81, 2022-03-09:
> Sadly they don't, (Umidigi Power and F1) If wireless charging is the "easy" fix, fine. But that information is missing from the main Github project page. An explanation or warning about this major [...]

You don't have to remove battery, just connect buck converter set to around 3.8-4V to battery terminals or use tp4056 charging module instead of buck converter. That way octo can run even if there's power disruption of some kind. Either option is good but buck is better in my opinion since it wont keep battery full all the time

Raul-7-7, 2022-03-09:
> Sadly they don't, (Umidigi Power and F1) If wireless charging is the "easy" fix, fine. But that information is missing from the main Github project page. An explanation or warning about this major [...]

I found its more a hit and miss, OTG + Charge (Y split cables) seem to work on some phones, I did try a few options on Samsung S7 but none worked, so I went the wireless direction, however, I seen many on Reddit reporting Y Split cable does charge and OTG for them. 
I am just grateful this project exists!!! saved me some $$$ and helps with environment.

Filo301, 2022-03-12:
I've had similar issue with xiaomi mi9. I'm using something like usbc dock with HDMI, ethernet etc. Solution was changing rom,  MIUI to pixel experience. No more issues so far! Thanks for great software

henfri, 2022-03-13:
Hello,
I also get no connection between my Ender and my S21.
The Charging is no prerequisite for the connection, right?

Best regards,
Hendrik

nickr1957, 2022-04-16:
Hi, just installed octo4a on my old Galaxy S9, connected a new OTG cable and plugged in.
I cannot see the printer at all. 
I tried going into Developer Options and setting USB mode to Android Auto but it cannot see the printer still.
What am I missing?

nickr1957, 2022-04-16:
I've installed on a Galaxy S9 but cannot see any printer. New OTG cable.
Any advice welcome.

miroan81, 2022-04-16:
> Hi, just installed octo4a on my old Galaxy S9, connected a new OTG cable and plugged in. I cannot see the printer at all. I tried going into Developer Options and setting USB mode to Android Auto but it cannot see the printer still. What am I missing?

Your cable is probably the problem, try connecting USB flash drive and see if its visible on your phone.

nickr1957, 2022-04-16:
Ues I can see files on a SD card plugged in using the same OTG cable

miroan81, 2022-04-16:
When you open oct4a app, do you see your printer there in connected printers?

nickr1957, 2022-04-16:
No, I see no printers, that's why I can't conndct.

miroan81, 2022-04-16:
And when you connect printer to a PC, can your PC see it?

nickr1957, 2022-04-16:
It works find from a PC using Print wit USB and from a Pi running Octoprint

miroan81, 2022-04-16:
Thats weird, I haven't had to change anything on my mobile to be able to recognize my printer. Few times I had a problem where I had to restart or fully shut down my mobile to open usb connection, so try that if you havent already.

nickr1957, 2022-04-16:
Its weirder than that! My phone just popped up a message out of the blue asking if I wanted to allow it to bridge using UART and when I said yes to that, my printer suddenly connected.
Didnt change any settings, didnt plug in the printer any differently (in fact it was plugged in for 10 mins or so already) so no idea why it suddenly asked for access...

nickr1957, 2022-04-16:
Whether it will work next time is a question but I think I'll try a print before disconnecting and reconnecting...

miroan81, 2022-04-16:
Thats good to hear, anytime you had something like that just restart both mobile and printer, it happens but rarely.  Im using this app for the last two months, works perfectly, I've had few week long prints with octolapse runing in backgound, it didn't fail so far. Just install arcwelder in Cura or whatever slicer you are using, it helps immensely with slow baudrate

nickr1957, 2022-04-16:
Arcwelder? Thanks, I'll take a look. Printing seems to be working, the crunch cones when I disconnect everything andctry to reconnect...

miroan81, 2022-04-16:
Yeah, its a plugin for slicer as well as octoprint that simplifies gcode so when printing complicated things with a lot of curves you dont get strange artifacts due to slow datarate and nozzle shutter. 
https://community.octoprint.org/t/noticeably-worse-quality-from-octoprint-than-from-sd-card-ender-3-v2/33797/2

alufers, 2022-10-07:
Hi, thanks for trying octo4a!

Please try connecting an USB mouse first. If a cursor doesn't appear this means this is a problem with the OTG cable or your phone does not support OTG.

Cheers!

PS: feel free to ping me if you need this issue to be re-opened

closed 2022-10-07 by alufers

nickr1957, 2022-10-07:
I use OTG for many other things - camera control, external keyboard, digital microscope. So I know OTG works.