feelfreelinux/octo4a #146: Just installed the Octoprint app...

dangel1967, 2022-01-26:
... everything went good except it says it's not connected to the printer. The printer is connected to the pi. How exactly are we supposed to connect and run this thing? Kinda lost and disappointed here.

bobafett81, 2022-01-28:
Same thing here, my printer connects to fluid with raspi but won't connect to octo4a on Samsung s9 even with data cables

razhklash, 2022-02-07:
You could provide information on what device you're using, what kind of cable you're connecting with, output from the log when you try to connect?

manningcustom, 2022-02-10:
I think I have the same issue.    I am trying to use octoprint4a.   I have it running on my tablet.   The server is running and it says the printer is connected.   But when I go to octoprint in my browser it will not connect and just  times out.   I have switched from auto to manual and tried every listed baudrate.   no luck

What am I missing?

alufers, 2022-10-06:
Hi, thanks for trying octo4a!

Please try connecting an USB mouse first. If a cursor doesn't appear this means this is a problem with the OTG cable or your phone does not support OTG.

Cheers!

PS: feel free to ping me if you need this issue to be re-opened

closed 2022-10-06 by alufers