feelfreelinux/octo4a #156: [My fault I guess] [Question] Printer needs to be un- and re-plugged after each reboot

mcdope, 2022-02-08:
Hi,

I'm having a kinda weird "issue", I need to unplug the printer after each device reboot and reconnect it. Else Octoprint won't find it / be able to connect. 

After I replug the printer it works like expected/a charm. I guess I messed some setting up?

My device: Cheapo china Android TV box, Bqeel U1 Pro (RK3318 Quad-Core 64bit Cortex-A53, 4GB RAM, Android 10)

Of course it's not really an issue, since I can either just keep the box running even when the printer is off or simply "do the usb shuffle". But would be nice if anyone has a hint to fix it.

auratum, 2022-03-16:
Hey there,
same here with a huawei mate 20 pro. I think its because charging/ power supply is evaluated as higher priority than data transfer.

auratum, 2022-03-18:
> For me it's caused by the permission dialog. I'm running my device without any display, so I never noticed till I connected the TV recently...
> 
> After startup there is a permission dialog waiting for the user, to allow the serial access - even though I ticked "remember" last time. It then works if I either confirm the dialog or disconnect the cable and reconnect it - and that point it seems the remember thingy kicks in.
> 
> I guess this is more an Android permission thingy then an actual issue/thingy in octo4a

You where totally right.
The phone switched back to charging mode. Fixed this to file transfer and now it works like charme. Thank you for your help.

jdmz1, 2022-08-05:
> For me it's caused by the permission dialog. I'm running my device without any display, so I never noticed till I connected the TV recently...
>
> After startup there is a permission dialog waiting for [...]

I was able to fix the permissions issue by recompiling the apk with a couple changes to the devices.xml file