feelfreelinux/octo4a #158: Installation corrupted... But with a twist.

Negat1veZer, 2022-02-11:
Trying to install octo4a on my moto g pure, however 
I get the serialpipe error and the installation corrupted. I allowed access to managing files, i've tried 1.0.3, and I've tried reinstalling the software half a dozen times, but alas twas all for not. Any help would be greatly appreciated.
[image]
[image]
[image]

Negat1veZer, 2022-02-11:
As another side note, the app crashes as soon as it finishes loading. When I open it again it has the installation corrupted screen.

miroan81, 2022-02-12:
Exactly the same problem on my Xperia z1, android 5.

lpka, 2022-02-17:
the same problem, with Andorid 7 and andorid 10 on Galaxy tab s 8.4. formated the device, new install. same problem.

miroan81, 2022-02-17:
Whats strange I also have lenovo with the same android 5 as z1, octo installed properly but the phone does not support OTG so I cant use that one.

miroan81, 2022-02-19:
Octo is running fine now on my Z1 after I flashed custom android 10 fw. Anyone else wants this to work, go to xxda forums, flash your device with something  esle, you have nothing to lose but your time :) Good luck!

stevvie, 2022-05-18:
Same ehere with Samsung Tab 4 with 1.1.1

pnmartinez, 2022-09-22:
Same here on Xperia Z3, Android 6.

HexRebuilt, 2023-01-14:
Same here with a Xiaomi Redmi pro, android 6

alufers, 2023-03-26:
Should be fixed in the newest version (1.2.4). Please open another issue if its not.

closed 2023-03-26 by alufers