feelfreelinux/octo4a #168: Camera resolution settings do not change (1.0.4)

atilacsf, 2022-03-03:
I'm using a Samsung A51 with octo4a 1.0.4 build. I notice that when I change the camera resolution it doesn't really change, no matter what resolution I set the output is always 1:1 and I suspect it is 1088x1088. I would like to lower the resolution, as I don't need tsuch a crisp image, but would like it to be smoother (low fps now with this settings).

Also, I can choose from 4 cameras (two front, two rear, although the phone has 4 rear and just one front cameras) but it seems it really doesn't change the camere, as both rear camera looks just the same.

Image rotation and turn on flash are working perfect.

Any ideas on how to solve this?

Thank you very much

kristoferahlqvist, 2022-10-25:
I have the exact same problems with my Huawei p20 pro

YoSarin, 2023-01-22:
Same issue here with Noika 6.
Once I briefly got 1080p, but can't get anywhere close to it now, both mjpeg stream and snapshot are stucked at 640x480.

Masi97, 2023-02-12:
Same with a Galaxy A8

brad, 2023-10-26:
I noticed the same issue with my aspect ratio in OctoPrint set to 16:9, but when I set the camera resolution to 1920 x 1080 (a valid 16:9 resolution) it started working

TobGrab, 2024-01-09:
Same on mine. Samsung A40 (original os)

DBShrill, 2024-12-01:
Same on mine with Huawei Mate 30.