feelfreelinux/octo4a #172: Flash Light toggle command

K3LS0, 2022-03-06:
Is there currently a command/script I can use with a custom Octoprint control button to toggle the phone's flashlight LED. I don't want to always have the light on when viewing the camera stream, however I would like the ability to toggle it on in low light scenarios.

Grover49, 2022-03-29:
I would be very interested in this as well.

Man-in-Black, 2022-06-30:
I would also appreciate it.
Also the LED switches off when I'm using another browser tab.

titanreign, 2022-10-03:
Another user interested in this feature!

DonEugraf, 2023-02-06:
If flash light would turn on itself when taking pictures by octolapse would solve black pictures and saves me having a laptop "watching" at all times.
This would be a great improvement.

Pabicchio, 2023-02-06:
This feature would be very useful

sanjiv204, 2024-01-25:
Very much needed for octo4a.

Pabicchio, 2024-01-25:
> If flash light would turn on itself when taking pictures by octolapse would solve black pictures and saves me having a laptop "watching" at all times.
> This would be a great improvement.

That would be amazing to have