feelfreelinux/octo4a #173: Unable to install it on Redmi 4A

dmajbs, 2022-03-07:
Hi, 

I am trying to install it on Redmi 4A (Miui 10.2.3, 2GB RAM, 4core 1.4Ghz) but I get this error message in the first step of instalation. Any idea how can I procede? Thank you so much for help!!

This is the error:

java.lang.RuntimeException: Unable to create directory: /data/data/com.octo4a/files/bootstrap-staging/root
	at com.octo4a.repository.BootstrapRepositoryImpl.ensureDirectoryExists(BootstrapRepository.kt:149)
	at com.octo4a.repository.BootstrapRepositoryImpl.access$ensureDirectoryExists(BootstrapRepository.kt:31)
	at com.octo4a.repository.BootstrapRepositoryImpl$setupBootstrap$2.invokeSuspend(BootstrapRepository.kt:96)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

portionDaOG, 2022-11-23:
got the same issue , at the same device , at the 1st screen after instaling the apk  , it doesnt have any button to proceed with it for the 700mb instalation

alufers, 2023-03-21:
Please try the new versions.

closed 2023-03-21 by alufers