feelfreelinux/octo4a #182: Webcam stream is very green

daitoZF, 2022-03-25:
Hi, my webcam is appearing a very bright green. I saw this issue elsewhere but it was closed and I saw that there was a patch for it a couple versions ago. I installed Octoprint yesterday and am on 1.0.4 . Any help would be appreciated
[image]
 :)

aarongoes, 2022-04-06:
This is a bug that has been fixed in Android 5.1.1 https://issuetracker.google.com/issues/37014456
What android version are you using?

BaSsAmMq, 2022-06-02:
I also have the same issue. 
Am using a Samsung Galaxy S4 Android version 5.0.2.
And am on the latest build of octa4a 1.1.1.

RLKnecht, 2022-06-16:
So since the Galaxy S4 is only 5.0.2 (my verizon version is actually only 5.0.1) I guess there is no fix available for us. I guess I can stop searching for a fix. So sad.

BaSsAmMq, 2022-06-17:
You are right. The only thing i might think of is to install a custom rom that has an android version that is more than 5.1.1, but the issue i have is that my phone is an exynos version that's why I don't have options, but if your phone is snapdragon then you can install lineageos from https://forum.xda-developers.com/t/official-lineageos-18-1-for-the-samsung-galaxy-s4-all-qcom-variants.4255313/
. Am not 100% sure if this will work but this will prevent you from throwing the phone 😅.