feelfreelinux/octo4a #183: Crashing during Install.

Superdad1106, 2022-03-26:
Hello. I was so excited to find your app. Raspberry pi's are quite price gouged at the moment and I went out and bought a new phone to use your app. I don't know anything about coding really outside of some basic HTML though but I have the apk installed on the phone and every time it starts to install the server or whatever the long install should be it will stay on that screen for maybe 15 seconds and then just crash and close out. I have tried google and all I found was try behind a vpn but the free one I tried didn't seem to do the trick. Any advice or help would be super appreciated. I will do my absolute best to get you any information you might need from me.

OzGav, 2022-04-11:
Try version 0.0.5 that at least installed for me

alufers, 2022-10-07:
Please try the latest version.

closed 2022-10-07 by alufers